sidoh/esp8266_milight_hub #121: Openhab integration breaking on 1.6.0 dev 5

DennisVonDerBey, 2017-10-04:
Hey guys,
just wanted to tell you that the dev 5 of version 1.6.0 broke my openhab integration. Didn't really have any time to analyze the problem but the state changes from openhab didn't do anything. The webui of the esp worked as usual. 
I have it setup as a V3 bridge.

If you need any more info I will help.

Thanks for your work again! :)

sidoh, 2017-10-04:
Thanks for the heads up.  I'll make sure it's working before releasing 1.6.  

I assume you're using the UDP integration?  1.6 has quite a few significant changes to the innards, so I probably broke something.

Serial logs with MILIGHT_UDP_DEBUG and DEBUG_PRINTF flags enabled would be very helpful.

garmck, 2017-10-05:
Just chiming in to say that same for Home Assistant, using IP with UDP ports V6. 
Fails with Connection Refused (other Python errors) 
Let me know if you want a separate issue

sidoh, 2017-10-05:
Was able to confirm UDP is broken in the 1.6-dev branches.  Please stick with v1.5 for now.

sidoh, 2017-10-05:
Should be fixed in 1.6.0-dev6:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.6.0-dev6

sidoh, 2017-10-07:
Gonna close, but please re-open if you notice any other issues.

closed 2017-10-07 by sidoh