sidoh/esp8266_milight_hub #127: device id

Nik0lov, 2017-10-08:
is there a way to get the device id without a remote or gateway? i think i cant sniff anything if i dont have a remote, right?

khmann, 2017-10-08:
The ID is only ever *learned* by the light bulb...  If you don't have one to transmit, make it up!  Out of the box most devices respond to "0000" but you really still need to pair with a real address for things to work right.

speendo, 2017-10-08:
you can set any ID you want (as long as it matches with the expected structure of IDs)

sidoh, 2017-10-08:
Thanks @khmann and @speendo.  This is exactly right -- use whatever ID between 0 and 0xFFFF (65535) you want.  Should probably add this to the README. :)

You'll need to pair new bulbs with whatever device ID you choose.  To do that, enter your device ID in the UI and select the appropriate remote type.  Depower the bulb (usually best to remove it from the socket), repower it, and within 5 seconds click on "pair" in the UI.

Going to close, but feel free to reopen if you have additional questions

> Out of the box most devices respond to "0000" 

Interesting, did not know that.

closed 2017-10-08 by sidoh

Nik0lov, 2017-10-08:
thanks a lot for the fast reply, but it doesnt work for me for some reason, 
i've got a FUT013 (5w rgbw e14)  it would not pair -_- have the ID to be entered in hex or dec system?

sidoh, 2017-10-08:
Just to confirm, the bulb is new and has not previously been paired with anything, correct?

You have RGBW selected here?

[image]

Can you enable sniffing before clicking on pair?  You should see something like this:

```
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 110C
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 0A
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 110C
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 09
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 110C
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 08
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 110C
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 07
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 110C
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 06
```

Nik0lov, 2017-10-08:
yeah the bulbs are complete new, never been paired with something, and yes, i selected RGBW
this is the sniff without pairing

> rgbw packet received (7 bytes):
Request type  : B0
Device ID     : AABB
b1            : 00
b2            : F9
b3            : 0E
Sequence Num. : 01
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : AABB
b1            : 00
b2            : B9
b3            : 0E
Sequence Num. : 00

sidoh, 2017-10-08:
Try moving the ESP/nRF really close to the bulbs.  10-20% of the nRF modules I've dealt with are either dead or have severely degraded range for some reason.  If you have another, might be worth swapping out.

Nik0lov, 2017-10-09:
I've ordered some new nrf, but if the sniffing tool shows packets, the nrf is Working, or not?

sidoh, 2017-10-09:
That's a good question -- locally sent packets are pushed through the same code path that intercepted packets are.  You'll see sent packets show up here no matter what.

Nik0lov, 2017-10-10:
holy moly @sidoh you were right, tried it with another nRF module and now it works :)) really crazy... next thing is connect it to pimatic, did someone tried this?? 
thanks a lot @sidoh awesome work!!!

sidoh, 2017-10-11:
Glad it's working for you, @Nik0lov.  I think there are a ton of clones of the nRF out there, and some of them have really horrendous quality.

I'm not really familiar with pimatic.  If the idea is to have the Pi drive the nRF directly, this code won't do that, as it's very tied to the ESP8266.  If you want to control the ESP8266 from your Pi, there are lots of ways to do that.  The REST API is probably the easiest.  MQTT is also supported.