sidoh/esp8266_milight_hub #128: Support the android App

donk7413, 2017-10-09:
Make the emulated gateway visible from the milight App on Android

starkillerOG, 2017-12-04:
@sidoh I am also having trouble with the android app and the UDP gateway servers.
On port 5987protocal version 5 the old V2 android app does see the ESP milight hub, but none of the buttons apear to work, the new V3 android app does not see the ESP milight hub.
On port 5987 protocal version 6 none of the apps see the ESP milight hub.

Are the UDP gateway servers compatible with having an admin username and pasword on the ESP milight hub, could that be the issue?

starkillerOG, 2017-12-04:
@sidoh tested some more and with the Bilight android app it did discover the ESP milight hub on protocal V6 and the buttons even caused pakets to be sent by the ESP, but the packets were diffrent than they schould be and therfore the bulb did not respond.

normal On command for group 4 rgbw bulb for me is:
b1: 4A
b2: C4
b3: 09
while if you press ON with group 4 selected and rgbw bulb selected in the Bilight app it gives this paket:
b1: 00
b2: 04
b3: 09
the Device ID and the type of packet were correct.

Maybe this gives some insight if someone is able to spent time debugging the UPD gateway servers.

druckgott, 2018-01-04:
I have the same problem

wknutsen, 2019-02-24:
I'm also having trouble with the milight app v3.
I did some sniffing on packets from the android app.
Seems like the Milight app is doing the discovery on port 5987 with this commands
`13 00 00 00 24 03 95 13 02 39 38 35 62 31 35 37 62 66 36 66 63 34 33 33 36 38 61 36 33 34 36 37 65 61 33 62 31 39 64 30 64`
and
`13 00 00 00 0a 03 95 13 11 f0 fe 6b e7 69 56 e3 09 35`
Cant find any documentation on command 13

sidoh, 2019-04-07:
Hey all, sorry I left you hanging here.

Honestly, this is probably not something I'm going to look into myself, but I'm happy to take a look at a PR.