sidoh/esp8266_milight_hub #129: Night Mode

vr6syncro, 2017-10-09:
Not a issue its a question.

At the moment i use the original Gateway and some not original Lights. Milight has the Night Mode. 
My Problem: it semms  that the Nightmode Brightness are set to low for my light. They are not lightning.

Is it possible to change Night Mode value with this alternativ Gateway? Or is this option stored on the Device?

Thank You

vr6syncro, 2017-10-10:
thanks for reply. 
with my gateway i only can push a button for brightness and i cant see the aktual value. is it available at this software to set a value and see the status?

sidoh, 2017-10-10:
Kind of.  You can give it a particular brightness value and it sets it by dimming the bulbs all the way (pressing down 10x), then pressing up the appropriate number of times.  Not completely reliable, but  works well enough.

vr6syncro, 2017-10-10:
ok that help me thanks

closed 2017-10-10 by vr6syncro