sidoh/esp8266_milight_hub #131: Cannot pair with RGBW bulb

jptrsn, 2017-10-11:
I'm not having any luck pairing with a newly-purchased MiLight RGBW. I've entered a device ID (0x0001 is the latest, but tried everything from 0x0000 through 0xFFFF) and I never see any indication on the bulb that it's paired.

Here's what the sniffer shows:
```
rgbw packet received (4 bytes):
Request type  : B0
Device ID     : 0001
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : 04
```

Am I correct in assuming that this means the NRF24L01 is connected properly? I don't have a remote for the bulb, so I can't test it or sniff any other traffic.

Help is greatly appreciated! Any suggestions, or troubleshooting? I'm not clear on what the `Gateway Servers` section is for.

Hardware is an ESP07 running v1.5.0. To reproduce the error, I power cycle the bulb, then hit the `pair` button, but not seeing any acknowledge on the part of the bulb

sidoh, 2017-10-11:
It likely means the nRF is connected correctly, but does not mean the nRF is working.  10-20% of nRFs I've dealt with don't work or have severely degraded range.  Try swapping in a new nRF.

If it's helpful, #127 is basically exactly the same issue, and swapping out the nRF fixed it.

jptrsn, 2017-10-11:
I looked at #127 and tested two different NRL24L01 modules, both are showing the same behaviour. I've used these same ones in other projects, so I know they're working, at least in the ~5 foot range in which I'm attempting to use it.

Any other suggestions of what to test?

sidoh, 2017-10-11:
Gotcha.

What's the model number of the bulbs you're testing with?  

If you want to 100% rule out a problem with the ESP, you can hook another one up and enable sniffing while the other is sending.

jptrsn, 2017-10-11:
It's listed on [ebay](http://www.ebay.ca/itm/RGBW-W-WW-E27-GU10-MR16-LED-Light-Dimmable-RGB-Bulb-Lamp-2-4G-Wireless-Milight/263136335485?ssPageName=STRK%3AMEBIDX%3AIT&var=562135287635&_trksid=p2057872.m2749.l2649) as an RGBWW (which identifies it as warm white, rather than cold white).

I'm wondering about the pull-down on GPIO15. I've tried using a different pin, but I'm seeing the same packets in the sniffer. Is CS active low? If so, then that's not the problem, but if it's active high, it may not be working properly on my board. I'll try flashing another ESP and sniff on that one while this one sends.

jptrsn, 2017-10-11:
I discovered the problem. It was related to not one, but two faulty power supplies that I'd tried. Now to get things set up properly with Home Assistant and the Android app. It seems that I can manage to use MQTT with Home Assistant, but I'm not clear on how to set the light back to white without going to the hub page and using the toggle there. I've used the example from the wiki, changed to match my RGBW bulb.

Thanks so much for building this! As a feature request, could you output debugging info to the serial pins on the ESP? I can see what I assume to be the WiFi Manager output, but nothing related to the operation of the radio or the packets.

closed 2017-10-11 by jptrsn

sidoh, 2017-10-11:
Glad you got it working!  Crazy how big a difference power supplies can make.  I've heard people sometimes have success adding a capacitor across Gnd and Vin.

You should be able to set the color back to white by selecting a color that's very close to white in the color picker.  Think there's also a way to do it if you enable effects on the mqtt_json component, but I don't have personal experience with that.  Some related discussion is in #106.

To keep things lightweight, debug output is disabled by default.  There are a couple of compiler flags you can add that will enable output:

https://github.com/sidoh/esp8266_milight_hub/blob/1.5.0/platformio.ini#L26