sidoh/esp8266_milight_hub #132: Lose the connection some times

donk7413, 2017-10-11:
The esp stop to send packet from udp after some times. I can't changes the color, or switch on/off. The blue led doesn't blink. I notice I can juste dim the light in RGB mode (RGBW can't) 
The version is the 1.6 dev7

sidoh, 2017-10-11:
Could you provide detailed steps that reproduce the issue you're referring to?

> I notice I can juste dim the light in RGB mode (RGBW can't)

Is this a separate issue?  RGBW definitely works.  I just tried it. :)

donk7413, 2017-10-11:
In fact it produce randomly, I have noticed it this morning. The only way to make it work again is to change the value of packet send. I will investigate this afternoon after work and try to know exactly how to reproduce it. I keep you in touch.

corneyl, 2017-10-12:
I already have problems with the UDP servers since 1.5, and it feels a bit like a memory problem. After a certain amount of packages it just stops working. From then on only one or two packages per minute result in messages to Milight bulbs. MQTT then still works perfect.

donk7413, 2017-10-12:
For I what 8 see, it's look like the wifi module crash. The led on the esp that usually blinks when command is send will stop blinking and always turn on. I don't know why and how to reproduce it. It's happen randomly.

sidoh, 2017-10-15:
I noticed when stress-testing the MQTT integration that sending response packets while under heavy load totally bunks the network stack (see #124).  I suspect this is because the ESP isn't able to clear internal buffers quickly enough, but I don't know enough about its guts to say for sure.

Are you guys using the V6 UDP server?  This one sends responses for every packet received, which is pretty likely to trigger this issue.  You might have more luck with V5 which does not have response packets, and is also waaaaaay more lightweight.

@corneyl did UDP work more reliably for you in a previous version?

corneyl, 2017-10-15:
I've not used a version before 1.5, but in this version it didn't work stable enough to use it. I think I also tried the v5 UDP server, but can't remember whether that worked better.

sidoh, 2017-10-15:
What kinds of stability problems did you notice, @corneyl?  I think there are people using it with OpenHAB, etc.  I'm guessing the problem is triggered by high (relative) volumes of packets.

Are the other integrations an option?  Both MQTT and REST are way more reliable simply by virtue of being over TCP.  Also working on some mildly fancy rate-limiting stuff for MQTT ( #124 ) which will improve performance under heavier loads.  If you use HomeAssistant, integrating is super duper easy with the `mqtt_json` component.

corneyl, 2017-10-15:
As stated above:
> After a certain amount of UDP packages it just stops working. From then only one or two UDP packages per minute result in messages to Milight bulbs.

So it feels like something, a buffer e.g., saturates. And sure, MQTT with HomeAssistant works perfect, so I'm using that without problems. But with UDP it would be possible to use transitions, that's why I looked into it at all.

sidoh, 2017-10-15:
Ah, yeah.  The limitless-led python library implements transitions by sending high volumes of UDP packets (not that there's really a better way).  UDP stuff can certainly be optimized, especially V6.  I think you'd probably have a lot more luck with V5 if you want to use transitions.  Packets are 2 bytes instead of like 30, and don't include responses.

Skinah, 2017-10-31:
Chris what update settings r u using ? To me I get no more loss of connections like earlier firmwares but it seems like the rf can not send the data to the globe as fast as the incoming mqtt can be received. In openhab I swipe around a colour wheel and the globes does change colours and appears not to drop any but one second of movement takes the globe ten seconds to catch up before the buffer is cleared and I can make new updates to the globe.

Previous firmwares if I spammed mqtt the globe would refuse to change at all and would not respond for around 20 seconds and then it would work as per normal.

So my guesses r it may be my mqtt broker or openhab not keeping up or it may be the rf is far slower due to the repeat settings or it may be a bug. I am away for a few days so I can’t run some tests to narrow the cause.

sidoh, 2017-11-01:
Think my settings are mostly the defaults (clipping things that aren't relevant):

```json
{
 "packet_repeats": 50,
  "http_repeat_factor": 1,
  "mqtt_topic_pattern": "milight2/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "",
  "mqtt_state_topic_pattern": "milight2/states/:device_id/:device_type/:group_id",
  "listen_repeats": 1,
  "state_flush_interval": 1100,
  "mqtt_state_rate_limit": 1300,
  "packet_repeat_throttle_sensitivity": 10,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 3
}
```

I would be shocked if OpenHAB or MQTT were the bottleneck.  If either of them are, there's probably some artificial rate limiting going on.

The last three settings are for a new feature I'll be merging shortly (see #141).

What I've been doing to narrow down the cause is:

1.  Set up a constant ping to the ESP.  If the network stack gets clogged you'll see lost packets.
2. Set up `mosquitto_sub` to listen on `milight2/#` or whatever your topic prefix is.  This way you can see the packets as they come in

I wrote this up last night.  Not sure how helpful it'll be:

https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning

sidoh, 2017-11-05:
What's in v1.6 RC1 contains several measures to improve performance.  You'll be best off with the MQTT integration.  V5 UDP should be okay too, but not as good.  V6 UDP will suck because it includes responses, which clogs up the network stack.

Feel free to re-open or open a different ticket if there are significant problems or anyone has suggestions.

closed 2017-11-05 by sidoh