sidoh/esp8266_milight_hub #133: Fire Disco Mode is white and not orange like the remote

donk7413, 2017-10-11:
When the Fire Disco Mode (simulation of a fire) is selected with a remote, it blinks in orange. With the ESP, it's in white.

sidoh, 2017-10-11:
What's the bulb model?

donk7413, 2017-10-11:
The model is RGBW e27 6w. I will send you the trace when I'm at home.

donk7413, 2017-10-11:
rgb_cct packet received (9 bytes):
Raw packet: EA C0 8A 5D A6 7C FD 6A EA 

Decoded:
Key      : EA
b1       : 20
ID       : 2B5F
Command  : 05
Argument : 00
Sequence : 00
Group    : 01
Checksum : 3E

sidoh, 2017-10-11:
Setting mode to 0 should have the same effect.  Does it not?

donk7413, 2017-10-11:
Not really, in fact the blinking is the same but the color is white, and the expected color should be orange.

sidoh, 2017-10-11:
That's a different mode.  Are you sure you're setting mode 0?  Could you paste the sniffed packet?

donk7413, 2017-10-11:
rgb_cct packet received (9 bytes):
Raw packet: 41 76 25 87 75 59 29 79 3A 

Decoded:
Key      : 41
b1       : 20
ID       : 2B5F
Command  : 05
Argument : 00
Sequence : 26
Group    : 02
Checksum : 50

donk7413, 2017-10-11:
This was the remote packet

sidoh, 2017-10-11:
That's a packet from your remote.  ESP will only send packets encoded with the key `00`.  Was hoping to see the sniffed packet sent from the ESP.

sidoh, 2017-10-11:
That's a totally different protocol and device ID.  Shouldn't you be using RGB+CCT and the device ID `2B5F`?

donk7413, 2017-10-11:
I have created virtual device and define the Id. I don't want use my remote Id. 
In fact my light doesn't work in RGB+CCT mode on the esp. It only work with RGBW.

donk7413, 2017-10-11:
That why I don't understand how the remote can send it in rgB CCT mode. My light are rgbw

sidoh, 2017-10-11:
I'm confused about your setup here.  You have a remote that sets mode 0 on a particular set of bulbs, and the ESP setting mode 0 does a different thing on the same set of bulbs?  Are the bulbs paired with both the remote and the ESP?

donk7413, 2017-10-11:
Sorry, i'm not clear. I will explain to you.
I have two bulb sync with the remote on the group 2.
And these bulb are sync on the esp too. On the group 1 of the device 0x03. I choose this device is by myself. Because I have 7 emulated gateway  (0x1, 0x2, ...) that I have created alone. My bulbs are on the the gateway 3 group 1.

sidoh, 2017-10-11:
If your bulbs work with the remote you have, then they work with the RGB+CCT protocol.  Mode 0 is probably interpreted differently between the two protocols.  Try syncing them with the RGB+CCT protocol on the ESP.

donk7413, 2017-10-11:
My bad. It was that. Everything run like hell now. Thanks again and sorry for the mistake.

sidoh, 2017-10-11:
Sweet, glad to hear it's working!

closed 2017-10-12 by sidoh