sidoh/esp8266_milight_hub #134: add to wiki

rhamblen, 2017-10-11:
Useful if you add one thing to wiki... (I couldnt see how to do it myself)

Integration to SMARTTHINGS
https://github.com/fireboy1919/MiThings

Ive been using this since I discovered you and it works perfectly.
Also it is now integrated with Alexa and works perfectly as well.

Just useful for people to link back to.

sidoh, 2017-10-11:
Sounds like a good idea.

I've not tried this myself, but I think you can contribute to the wiki by forking/cloning https://github.com/sidoh/esp8266_milight_hub.wiki.git.

Lstt2005, 2017-10-12:
Very interesting catalogs - compatibles and other information if 2 files:
[MiLight-ProductCatalog-2017_other.pdf](https://github.com/sidoh/esp8266_milight_hub/files/1379130/MiLight-ProductCatalog-2017_other.pdf)
[MiLight-ProductCatalog-2017.pdf](https://github.com/sidoh/esp8266_milight_hub/files/1379131/MiLight-ProductCatalog-2017.pdf)

sidoh, 2017-10-29:
Added a link to @fireboy1919's repo to the wiki.  Also added the product catalog.

closed 2017-10-29 by sidoh