sidoh/esp8266_milight_hub #138: lack of electricity

bonz88, 2017-10-18:
Whenever there is lack of electricity, when it comes back, NodeMCU won’t connect to the WiFi. I have to reconnect it manually. I don’t know where’s the problem, I guess that after the power goes on, the ESP is faster than the router, so it’s searching for a router to connect to, and it doesn’t exist (yet). Is there a way to change it in settings? Or add something in the code so that it waits a bit longer after being turned on?

hencou, 2017-10-18:
I experienced the same problem and solved it by replacing wifiManager and MQTT functions by https://github.com/PhracturedBlue/ESP8266MQTTMesh

sidoh, 2017-10-18:
Ah, thanks for catching this.  I think you're exactly right -- ESP boots before the router, and when it doesn't find a network to connect to, it opens the configuration AP.  You could just connect to this and save your settings again, but that's sort of cumbersome.

I had to solve the same issue for another project.  Should be as simple as 

```c++
wifiManager.setConfigPortalTimeout(180);
```

Which will cause the ESP to reboot after the timeout is reached.

closed 2017-10-29 by sidoh