sidoh/esp8266_milight_hub #142: stack trace when issuing commands

miknuc, 2017-10-31:
Hi,

I ran into a stack trace when trying to send a command to one of my milight bulbs.

What I did:
- download of https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.6.0-dev9 as source
- compile it using platformio (Linux/atom). I had to change platform into espressif8266 to perform a build
- upload to a nodemcu dev board (clone from lolin), with NFR board connected
- configure wifi connection upon first startup
- open web page, start sniffing mode
- send some commands to my bulbs using existing milight gateway
- sniffing showed device ID B516, which is correct (indication for me that the NFR connection seems to work)
- enter 0xb516 as Device ID (btw: why do you require the 0x prefix). Change colour or anything else
- boom, esp8266 reboots

This is the translated stack trace (using EspStackTraceDecoder):

Exception Cause: Not found

0x40104e5e: spi_flash_read at ??:?
0x40222f5b: pp_attach at ??:?
0x4021d95c: RF24::csn(bool) at ??:?
0x401073cc: delayMicroseconds at ??:?
0x4021d9b0: RF24::endTransaction() at ??:?
0x4021d95c: RF24::csn(bool) at ??:?
0x4021d99c: RF24::beginTransaction() at ??:?
0x4021dbfb: RF24::spiTrans(unsigned char) at ??:?
0x4021dc54: RF24::get_status() at ??:?
0x4021dd72: RF24::write(void const*, unsigned char, bool) at ??:?
0x4021ddb7: RF24::write(void const*, unsigned char) at ??:?
0x40212366: PL1167_nRF24::transmit(unsigned char) at ??:?
0x4010053d: realloc at ??:?
0x40211fa1: NRF24MiLightRadio::resend() at ??:?
0x40211f58: NRF24MiLightRadio::write(unsigned char*, unsigned int) at ??:?
0x40213ded: MiLightClient::write(unsigned char*) at ??:?
0x40211f30: NRF24MiLightRadio::write(unsigned char*, unsigned int) at ??:?
0x4021570d: RgbwPacketFormatter::command(unsigned char, unsigned char) at ??:?
0x40213e4a: MiLightClient::flushPacket() at ??:?
0x40213ec0: MiLightClient::updateHue(unsigned short) at ??:?
0x4021455d: MiLightClient::update(ArduinoJson::JsonObject const&) at ??:?
0x4021f428: MiLightClient::switchRadio(unsigned int) at ??:?
0x40213d58: MiLightClient::switchRadio(MiLightRemoteConfig const*) at ??:?
0x4021782e: MiLightHttpServer::handleRequest(ArduinoJson::JsonObject const&) at ??:?
0x40218600: MiLightHttpServer::handleUpdateGroup(UrlTokenBindings const*) at ??:?
0x40202256: String::String(char const*) at ??:?
0x402184a1: MiLightHttpServer::handleUpdateGroup(UrlTokenBindings const*) at ??:?
0x4010053d: realloc at ??:?
0x4010053d: realloc at ??:?
0x4021be1b: SPIClass::setDataMode(unsigned char) at ??:?
0x4020237d: String::operator=(char const*) at ??:?
0x40100688: free at ??:?
0x40202180: String::changeBuffer(unsigned int) at ??:?
0x402186b8: std::_Function_handler<void (UrlTokenBindings*), MiLightHttpServer::begin()::{lambda(UrlTokenBindings const*)#7}>::_M_invoke(std::_Any_data const&, UrlTokenBindings*) at MiLightHttpServer.cpp:?
0x40218c98: PatternHandler::handle(ESP8266WebServer&, HTTPMethod, String) at ??:?
0x40218c3e: PatternHandler::handle(ESP8266WebServer&, HTTPMethod, String) at ??:?
0x4020212f: String::~String() at ??:?
0x40202205: String::copy(char const*, unsigned int) at ??:?
0x4020230c: String::operator=(String const&) at ??:?
0x40218c08: PatternHandler::handle(ESP8266WebServer&, HTTPMethod, String) at ??:?
0x4020cae9: ESP8266WebServer::_handleRequest() at ??:?
0x4020979c: WiFiClient::~WiFiClient() at ??:?
0x402189ef: WebServer::_handleRequest() at ??:?
0x40218ace: WebServer::handleClient() at ??:?
0x40217360: MiLightHttpServer::handleClient() at ??:?
0x4022138a: loop at ??:?
0x40202da8: loop_wrapper() at core_esp8266_main.cpp:?
0x40204bd8: cont_norm at cont.o:?

Do you have any suggestions?

sidoh, 2017-10-31:
Can you try using one of the pre-compiled binaries and see if you have the same problem?  I can send commands to 0xb516 just fine, which almost certainly means one of our compilers (and the Travis.ci compiler if it's mine) is doing something weird, or you've got bad hardware.  It's definitely the case that the nRF can work for listening, but not sending.  Seen it a few times before.

> compile it using platformio (Linux/atom). I had to change platform into espressif8266 to perform a build

run this:

```
platformio platform install 'https://github.com/platformio/platform-espressif8266.git#feature/stage'
```

Trying to find a way to make this unnecessary, but platformio isn't making it easy.  Building against the RC version of the SDK is important because the stable version doesn't include the fix for KRACK.

> why do you require the 0x prefix

Because you can input a decimal value too.  I'd much rather be explicit and verbose than have weird unexpected issues. :)

miknuc, 2017-10-31:
Thank you for this ultra quick reaction :)
I tried 
https://github.com/sidoh/esp8266_milight_hub/releases/download/1.6.0-dev9/esp8266_milight_hub_nodemcuv2-1.6.0-dev9.bin

and got a similar reboot.
I attach the stack trace for your information although I don't know if the debug symbols of your binary are identical with the output of my own built:

ms@oscar2:~/Projekte/esp/platformIO/Projects/esp8266_milight_hub-1.6.0-dev9/.pioenvs/nodemcuv2$ java -jar ~/Projekte/esp/EspStackTraceDecoder.jar /home/ms/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-addr2line ./firmware.elf ~/tmp/dump2.txt 
Exception Cause: Not found

0x4021e948: Print::write(unsigned char const*, unsigned int) at ??:?
0x401064a8: __muldf3 at /home/igrokhotkov/xtensa/crosstool-NG/.build/src/gcc-4.8.2/libgcc/config/xtensa/ieee754-df.S:1170
0x401065a1: __divdf3 at /home/igrokhotkov/xtensa/crosstool-NG/.build/src/gcc-4.8.2/libgcc/config/xtensa/ieee754-df.S:1352
0x4021e948: Print::write(unsigned char const*, unsigned int) at ??:?
0x4021e99c: Print::print(__FlashStringHelper const*) at ??:?
0x4021ebf6: strncpy_P(char*, char const*, unsigned int) at ??:?
0x4021ec40: std::_Sp_counted_ptr<SPIFFSImpl*, (__gnu_cxx::_Lock_policy)0>::_M_get_deleter(std::type_info const&) at ??:?
0x4021ed72: spiffs_cache_page_free at spiffs_cache.c:?
0x4021edb7: spiffs_cache_page_allocate at spiffs_cache.c:?
0x40212ec6: GroupState::setSaturation(unsigned char) at ??:?
0x4010053d: realloc at ??:?
0x40212b05: LT8900MiLightRadio::read(unsigned char*, unsigned int&) at ??:?
0x40212abc: LT8900MiLightRadio::iReadRXBuffer(unsigned char*, unsigned int) at ??:?
0x4021492d: MiLightRemoteConfig::fromType(String const&) at ??:?
0x40212a98: LT8900MiLightRadio::iReadRXBuffer(unsigned char*, unsigned int) at ??:?
0x40216249: BulbStateUpdater::flushGroup(BulbId, GroupState&) at ??:?
0x4021498a: MiLightRemoteConfig::fromType(String const&) at ??:?
0x40214a00: MiLightRemoteConfig::fromType(MiLightRemoteType) at ??:?
0x4021509d: RgbCctPacketFormatter::parsePacket(unsigned char const*, ArduinoJson::JsonObject&, GroupStateStore*) at ??:?
0x40220300: LinkedList<BulbId>::remove(int) at ??:?
0x40214898: MiLightClient::update(ArduinoJson::JsonObject const&) at ??:?
0x4021b38e: V6MiLightUdpServer::handleCommand(unsigned short, unsigned char, unsigned char*, unsigned char, unsigned char) at ??:?
0x4021c174: PubSubClient::readByte(unsigned char*) at ??:?
0x402022da: String::move(String&) at ??:?
0x4021c020: SPIClass::setFrequency(unsigned int) at ??:?
0x4010053d: realloc at ??:?
0x4010053d: realloc at ??:?
0x40101ef8: wDev_ProcessFiq at ??:?
0x402023fd: String::operator=(__FlashStringHelper const*) at ??:?
0x40100688: free at ??:?
0x4020220f: String::copy(char const*, unsigned int) at ??:?
0x4021c22c: PubSubClient::readPacket(unsigned char*) at ??:?
0x4021c808: PubSubClient::loop() at ??:?
0x4021c7ae: PubSubClient::loop() at ??:?
0x4020222f: String::String(char const*) at ??:?
0x4020228d: String::copy(__FlashStringHelper const*, unsigned int) at ??:?
0x40202390: String::operator=(char const*) at ??:?
0x4021c778: PubSubClient::loop() at ??:?
0x4020d4cd: ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int) at ??:?
0x40209fdc: WiFiUDP::begin(unsigned short) at ??:?
0x4021c563: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at ??:?
0x4021c642: PubSubClient::publish(char const*, unsigned char const*, unsigned int, unsigned char) at ??:?
0x4021aec4: V5MiLightUdpServer::handleCommand(unsigned char, unsigned char) at ??:?
0x402221e2: ppPeocessRxPktHdr at ??:?
0x40202de4: yield at ??:?
0x40204d7c: spiffs_read_dir_v at spiffs_hydrogen.c:?

As a next step to exclude any hardware issues I will replace the esp8266 and the NFR board. I'll let you know about the outcome.

miknuc, 2017-10-31:
It was a hardware issue :(
Sorry for the confusion, I had a wiring issue, CE was on the wrong pin.
Now it works, the version built on my own system also. 
Thank you for the quick support!

sidoh, 2017-10-31:
Nice, glad it's working!  Thanks for following up, and for providing all of the information you did upfront.  Saved a lot of back and forth :)  

I'd like to write a "troubleshooting" section, and your notes will be helpful for that.

closed 2017-10-31 by sidoh