sidoh/esp8266_milight_hub #145: Opening settings file failed

bhaap, 2017-11-02:
Hi

Im getting this issue when trying to run your code.

``
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 172.16.138.86
Opening settings file failed
Setup complete
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
Opening settings file failed
``

bhaap, 2017-11-02:
If it helps im using a esp-03 module with 512KB of flashable mem

sidoh, 2017-11-02:
How are you flashing it?  Did you build the firmware yourself, or are you using one of the pre-compiled binaries?

bhaap, 2017-11-02:
This was with a precompiled one for esp07. 
When compiling it from source after editing the platform.ini to have my 512k64k ESP setting I get it to compile and upload but the device never shows up in WiFi and the serial monitor only shows random things at 9600 8-1.

sidoh, 2017-11-02:
I think the bootloader's default baudrate is 115200.

Are you sure your chip has 512 KB flash?  I think even newer esp-01s have 4 MB.

bhaap, 2017-11-02:
I will check the memory module on the board. But im pretty shure.

bhaap, 2017-11-03:
Yes its a 4M bit module (512kb)

The issue might be that its to small mem module. I think that one can only use about 423kb of memory. 
according to this blogpost atleast: https://delog.wordpress.com/2016/05/13/getting-started-with-the-esp-03/

Gonna try to upgrade the memory and reflash :)

sidoh, 2017-11-05:
Aah, this makes sense.  Looks like the binary is something like 470 KB right now, so makes sense that it wouldn't work if 423 KB is the limit.

bhaap, 2017-11-17:
Uploaded the latest release via ATOM to a Wemos d1 mini. Works like a charm.

Only thing i miss is the on/off transition.

closed 2017-11-17 by bhaap