sidoh/esp8266_milight_hub #146: No way to delete unused ID's from drop down list

RvonMallesch, 2017-11-02:
Hi,

Is there a way to delete an ID from the drop down list? Ideally, if an ID was no longer linked to any bulbs it would disappear.

Regards,

Richard

sidoh, 2017-11-02:
There isn't currently a way through the UI, but you can set the list to whatever you want via the REST API:

```
curl -v -X PUT -d '{"device_ids":[1,2,3]}' http://milight-hub.local/settings
```

There's no way for this to happen automatically, though.  There's no way to determine the number of bulbs linked to a particular ID.

RvonMallesch, 2017-11-02:
OK thanks :-)

sidoh, 2017-11-03:
Want to make sure I'm understanding your suggestion.  Were you asking if there's a way to do it in the UI?  That's totally reasonable and should probably be a thing.  I also think it'd be nice if the IDs in the dropdown auto-saved (they are currently only persisted when you save settings).

servim, 2017-11-25:
and if the identifier is only one, you need it to be selected automatically :)

sidoh, 2017-11-25:
Fair point.  Perhaps the most general way to handle that is to have the browser remember the last selected ID via local storage.

closed 2019-05-28 by sidoh