sidoh/esp8266_milight_hub #147: Request: Method to directly select a mode

RvonMallesch, 2017-11-03:
Hi again,

I believe it would be a great feature to have method to choose the disco mode directly. As you would know, to do this in the UDP command, the first command after the remote style is 0x06 and the following command repeated 4 times is the mode number.

Many thanks for your fantastic work on this project!

Richard

sidoh, 2017-11-03:
Which bulb type are you wanting to do this for?  It's only not possible with RGB, and this is because my understanding of the protocol has led me to believe it's not possible.

You should see this in the UI for RGBW, RGB+CCT, and FUT089:

[image]

It's possible to set the mode directly for these bulb types in the REST API or MQTT integration by supplying, for example, `{"mode":5}` in the request body or message payload.

Is this not what you mean?

RvonMallesch, 2017-11-03:
DOH! Please ignore me, I simply failed to see it in the list of commands available.

Apologies for wasting your time on this,

Richard

sidoh, 2017-11-03:
No worries -- the documentation isn't exactly perfectly organized either.  Glad you found what you're looking for :)

closed 2017-11-03 by sidoh