sidoh/esp8266_milight_hub #148: Openhab integration

Tertiush, 2017-11-04:
First off an awesome project, thanks for taking it on and keeping it alive!!

I'm using Openhab2 and have three existing V3 bridges with a lot of bulbs synced. I've used your hub code to successfully simulate two bridges and can control the lights from your web interface. My problem comes when bringing Openhab into the mix....

A single v5 bridge works fine (in OH2 I used the v3 milight bridge channel). A second v5 simply doesnt work as openhab doesn't even send any packets to the bridge, odd, but then I moved to v6. V6 does in fact send packets to the virtual gateways and I can see replies, but for some reason the gateways is not relaying the commands to the bulbs. Below is a tcpdump of the traffic when sending an ON and OFF respectively to each simulated bridge. Any ideas what could cause this? thanks

The setup is:
OH2 on RPI: 10.0.0.200
ESP Bridge IP: 10.0.0.208
Gateway 1: ID: 0x5E4D, port 3389, v6
Gateway 2: ID: 0x7B89, port 3390, v6

**:~$ sudo tcpdump -v -X -n host 10.0.0.208**
tcpdump: listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
16:08:48.789192 IP (tos 0x0, ttl 64, id 16349, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 7
        0x0000:  4500 0023 3fdd 4000 4011 e555 0a00 00c8  E..#?.@.@..U....
        0x0010:  0a00 00d0 d943 0d3d 000f 15b8 d000 0000  .....C.=........
        0x0020:  0200 00                                  ...
16:08:48.791785 IP (tos 0x0, ttl 64, id 16350, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.41786 > 10.0.0.208.3390**: UDP, length 7
        0x0000:  4500 0023 3fde 4000 4011 e554 0a00 00c8  E..#?.@.@..T....
        0x0010:  0a00 00d0 a33a 0d3e 000f 15b8 d000 0000  .....:.>........
        0x0020:  0200 00                                  ...
16:08:48.797959 IP (tos 0x0, ttl 128, id 5743, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3389 > 10.0.0.200.55619**: UDP, length 12
        0x0000:  4500 0028 166f 0000 8011 0ebf 0a00 00d0  E..(.o..........
        0x0010:  0a00 00c8 0d3d d943 0014 d74e d800 0000  .....=.C...N....
        0x0020:  0700 0000 005e 4d00 0000 0000 0000       .....^M.......
16:08:48.805204 IP (tos 0x0, ttl 128, id 5744, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3390 > 10.0.0.200.41786**: UDP, length 12
        0x0000:  4500 0028 1670 0000 8011 0ebe 0a00 00d0  E..(.p..........
        0x0010:  0a00 00c8 0d3e a33a 0014 d139 d800 0000  .....>.:...9....
        0x0020:  0700 0000 007b 8900 0000 0000 0000       .....{........
16:08:48.889720 IP (tos 0x0, ttl 64, id 16360, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 7
        0x0000:  4500 0023 3fe8 4000 4011 e54a 0a00 00c8  E..#?.@.@..J....
        0x0010:  0a00 00d0 d943 0d3d 000f 15b8 d000 0000  .....C.=........
        0x0020:  0200 00                                  ...
16:08:48.915803 IP (tos 0x0, ttl 128, id 5745, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3389 > 10.0.0.200.55619**: UDP, length 12
        0x0000:  4500 0028 1671 0000 8011 0ebd 0a00 00d0  E..(.q..........
        0x0010:  0a00 00c8 0d3d d943 0014 d74e d800 0000  .....=.C...N....
        0x0020:  0700 0000 005e 4d00 0000 0000 0000       .....^M.......
16:08:49.418224 IP (tos 0x0, ttl 64, id 16361, offset 0, flags [DF], proto UDP (17), length 31)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 3
        0x0000:  4500 001f 3fe9 4000 4011 e54d 0a00 00c8  E...?.@.@..M....
        0x0010:  0a00 00d0 d943 0d3d 000b 15b4 4500 55    .....C.=....E.U
16:08:50.140526 IP (tos 0x0, ttl 64, id 16363, offset 0, flags [DF], proto UDP (17), length 31)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 3
        0x0000:  4500 001f 3feb 4000 4011 e54b 0a00 00c8  E...?.@.@..K....
        0x0010:  0a00 00d0 d943 0d3d 000b 15b4 4600 55    .....C.=....F.U
16:08:51.179473 IP (tos 0x0, ttl 64, id 16439, offset 0, flags [DF], proto UDP (17), length 31)
    **10.0.0.200.41786 > 10.0.0.208.3390**: UDP, length 3
        0x0000:  4500 001f 4037 4000 4011 e4ff 0a00 00c8  E...@7@.@.......
        0x0010:  0a00 00d0 a33a 0d3e 000b 15b4 4b00 55    .....:.>....K.U
16:08:51.802600 IP (tos 0x0, ttl 64, id 16476, offset 0, flags [DF], proto UDP (17), length 31)
    **10.0.0.200.41786 > 10.0.0.208.3390**: UDP, length 3
        0x0000:  4500 001f 405c 4000 4011 e4da 0a00 00c8  E...@\@.@.......
        0x0010:  0a00 00d0 a33a 0d3e 000b 15b4 4c00 55    .....:.>....L.U
16:08:53.791174 IP (tos 0x0, ttl 64, id 16665, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.41786 > 10.0.0.208.3390**: UDP, length 7
        0x0000:  4500 0023 4119 4000 4011 e419 0a00 00c8  E..#A.@.@.......
        0x0010:  0a00 00d0 a33a 0d3e 000f 15b8 d000 0000  .....:.>........
        0x0020:  0200 00                                  ...
16:08:53.791455 IP (tos 0x0, ttl 64, id 16666, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 7
        0x0000:  4500 0023 411a 4000 4011 e418 0a00 00c8  E..#A.@.@.......
        0x0010:  0a00 00d0 d943 0d3d 000f 15b8 d000 0000  .....C.=........
        0x0020:  0200 00                                  ...
16:08:53.797302 IP (tos 0x0, ttl 128, id 5746, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3389 > 10.0.0.200.55619**: UDP, length 12
        0x0000:  4500 0028 1672 0000 8011 0ebc 0a00 00d0  E..(.r..........
        0x0010:  0a00 00c8 0d3d d943 0014 d74e d800 0000  .....=.C...N....
        0x0020:  0700 0000 005e 4d00 0000 0000 0000       .....^M.......
16:08:53.800725 IP (tos 0x0, ttl 128, id 5747, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3390 > 10.0.0.200.41786**: UDP, length 12
        0x0000:  4500 0028 1673 0000 8011 0ebb 0a00 00d0  E..(.s..........
        0x0010:  0a00 00c8 0d3e a33a 0014 d139 d800 0000  .....>.:...9....
        0x0020:  0700 0000 007b 8900 0000 0000 0000       .....{........
16:08:58.791652 IP (tos 0x0, ttl 64, id 17012, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.41786 > 10.0.0.208.3390**: UDP, length 7
        0x0000:  4500 0023 4274 4000 4011 e2be 0a00 00c8  E..#Bt@.@.......
        0x0010:  0a00 00d0 a33a 0d3e 000f 15b8 d000 0000  .....:.>........
        0x0020:  0200 00                                  ...
16:08:58.792605 IP (tos 0x0, ttl 64, id 17013, offset 0, flags [DF], proto UDP (17), length 35)
    **10.0.0.200.55619 > 10.0.0.208.3389**: UDP, length 7
        0x0000:  4500 0023 4275 4000 4011 e2bd 0a00 00c8  E..#Bu@.@.......
        0x0010:  0a00 00d0 d943 0d3d 000f 15b8 d000 0000  .....C.=........
        0x0020:  0200 00                                  ...
16:08:58.798702 IP (tos 0x0, ttl 128, id 5748, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3389 > 10.0.0.200.55619**: UDP, length 12
        0x0000:  4500 0028 1674 0000 8011 0eba 0a00 00d0  E..(.t..........
        0x0010:  0a00 00c8 0d3d d943 0014 d74e d800 0000  .....=.C...N....
        0x0020:  0700 0000 005e 4d00 0000 0000 0000       .....^M.......
16:08:58.803903 IP (tos 0x0, ttl 128, id 5749, offset 0, flags [none], proto UDP (17), length 40)
    **10.0.0.208.3390 > 10.0.0.200.41786**: UDP, length 12
        0x0000:  4500 0028 1675 0000 8011 0eb9 0a00 00d0  E..(.u..........
        0x0010:  0a00 00c8 0d3e a33a 0014 d139 d800 0000  .....>.:...9....
        0x0020:  0700 0000 007b 8900 0000 0000 0000       .....{........

sidoh, 2017-11-04:
Thanks for providing the packet capture.  Because it looks like you have packets coming back, I would expect that the ESP is sending whatever RF packets it deems appropriate given what it parses from the UDP packet.

Could you try compiling with the `UDP_DEBUG` flag turned on (in `platformio.ini`)?  This will enable some debug messaging over Serial that'll probably be helpful.

sidoh, 2017-11-04:
V6 UDP should definitely work, but I'd strongly recommend checking out the MQTT integration if at all possible. 

A kind user on the OpenHAB community forums was kind enough to write up instructions on using the MQTT integration: 

https://community.openhab.org/t/use-esp8266-to-control-unlimited-groups-of-milight-or-limitless-led-bulbs-in-openhab2/34906

Tertiush, 2017-11-05:
Thanks for the quick reply! I have looked at the MQTT option before, but my rules in OH2 is all based off the single "color" item that represents all methods available for the bulbs. Discrete definitions for on/off, saturation, hue, etc. would be a bit of a nightmare for me to manage and keep track of in OH...

I got my PlatformIO environment setup and enabled the debug flag (just a note the current extra_script location for the flag didn't work, so I moved it to the build options section as: -D MILIGHT_UDP_DEBUG=1)

Below is the debug output of this, I bolded the "important" packets. The switching sequence from OH2 is the same as before (ON and OFF for each of the two bulbs, on separate gateways).

Hope this help, please let me know if I should try anything else. I'm also more than willing to contribute - I have a lot of experience reverse engineering protocols ;)

Received request with untracked session ID: 0
[MiLightUdpServer port 3390] - Handling packet: D0 00 00 00 02 00 01
Packet size: 1073683680
Received request with untracked session ID: 1
[MiLightUdpServer port 3390] - Handling packet: D0 00 00 00 02 00 01
Packet size: 1073683680
Received request with untracked session ID: 1
**[MiLightUdpServer port 3389] - Handling packet: 45 00 55
Packet size: 1073683680
Unhandled V6 packet
[MiLightUdpServer port 3389] - Handling packet: 46 00 55
Packet size: 1073683680
Unhandled V6 packet
[MiLightUdpServer port 3390] - Handling packet: 4B 00 55
Packet size: 1073683680
Unhandled V6 packet**
[MiLightUdpServer port 3389] - Handling packet: D0 00 00 00 02 00 00
Packet size: 1073683680
Received request with untracked session ID: 0
[MiLightUdpServer port 3390] - Handling packet: D0 00 00 00 02 00 01
Packet size: 1073683680
Received request with untracked session ID: 1
[MiLightUdpServer port 3389] - Handling packet: D0 00 00 00 02 00 00
Packet size: 1073683680
Received request with untracked session ID: 0
**[MiLightUdpServer port 3390] - Handling packet: 4C 00 55
Packet size: 1073683680
Unhandled V6 packet**
[MiLightUdpServer port 3390] - Handling packet: D0 00 00 00 02 00 01
Packet size: 1073683680
Received request with untracked session ID: 1

Tertiush, 2017-11-05:
Its working!!!! Turns out my understanding of how the OH binding works was a tad off, I think... Although configured as a V6 bridge, I had to get the correct bulb version. Strangely enough the same bulb version (rgbLed) I used for the V3 bridge didn't work on the V6 gateway. So I tried an apparent "later" (rgbwLed) bulb version and viola.

Once I have my setup done I'll write-up a little howto on the OH forums for others to use. This now means your project is a direct replacement for my 3 bridges with minimal changes needed in OH.

Thanks again for all the help and your effort towards this amazing project!! If you ever decide to take donations I for one will give back! A little paypal donate button ;)

cheers

closed 2017-11-05 by Tertiush

Tertiush, 2017-11-06:
@sidoh 

If you want to, I've made a little how-to for OH2 users using the latter V2 MiLight binding. I've successfully integrated all three bridges and its working like a charm, very impressed! Here's a link if you want to post it somewhere: https://community.openhab.org/t/the-new-milight-ibox-and-milight-binding/17621/275

sidoh, 2017-11-06:
Awesome!  Thanks for taking the time to write this up.

Added a page on the wiki linking to your post: https://github.com/sidoh/esp8266_milight_hub/wiki/Integrating-with-OpenHAB

Skinah, 2018-02-05:
Just a heads up, I am probably a week away from having a working Openhab2 binding for this EspMilightHub project. It auto finds and adds the globes to openhab making setup a breeze and also supports the ability for a milight remote to update Openhabs controls to name a few features.