sidoh/esp8266_milight_hub #152: Always same Device ID

vr6syncro, 2017-11-10:
I turned on my devices and the sniffer and pushed on all 3 lamps a button. The sniffer shows the command but the device id is always the same.
cct packet received (7 bytes):
Request type  : 5A
Device ID     : 1090

Whats wrong?

Skinah, 2017-11-10:
I don't see anything wrong with that unless u give more details.
Every button on the same remote will send the same device ID. The device ID is perhaps better thought of as the "Device ID of the remote".

Different remotes should have a different ID. 
The globes do not have any ID because they do not transmit RF.

vr6syncro, 2017-11-10:
oh ok i misunderstood that. Ok then its the id of my remote.
i added the id and saved a server. after that have i to link them again with the new hub or is it now finished?

Skinah, 2017-11-10:
That depends.

If you setup a server with the same Device ID as your remote, AND the remote is paired to globes then you should be good to go.

If you setup a new server with a DIFFERENT device id to your remote you will then need to pair/sync the ESPMH with the globes using the new device ID.

You only setup the servers if using UDP, that is not needed for MQTT.

sidoh, 2017-11-11:
Thanks @Skinah :)

Skinah is right.  A single physical remote has one Device ID.  The different buttons correspond to a Group ID.

ESPMH can spoof your remote.   Just enter the appropriate Device ID, select which Group ID you want to control, and make sure to select the relevant remote type (looks like it should be CCT):

[image]

Then you'll be able to control through the UI.

vr6syncro, 2017-11-11:
Ok Thank you. But One more question. Is the Connection then over the Remote? Or with the device itself? Its a Bit confusing

sidoh, 2017-11-11:
ESPMH is emulating the RF protocol used by the remote.  You don't need to use the remote at all if you don't want to.

sidoh, 2017-11-11:
Going to close, but please feel free to continue discussion here if you have more questions.

closed 2017-11-11 by sidoh