sidoh/esp8266_milight_hub #155: ESP crash after dimming CCT

mcguiresean, 2017-11-11:
I have an issue whereby once I have issued a dim command to any CCT bulb, the ESP crashes the next time I issue a command using RGBW.
If I use basic on/off commands, I can switch between the two types indefinitely, but once I have issued a dim command in CCT mode, the next command I send in RGBW mode will cause a reboot.

I have used NodeMCUs, Wemos D1 Minis and both NRF24 and LT8900 but the problem persists.

Currently using 1.6.0-rc3 but I have experienced this in all 1.6 versions. 1.5 does not seem to be affected.

sidoh, 2017-11-11:
Confirmed I'm seeing this too. I'll take a look.

sidoh, 2017-11-11:
Fixed in https://github.com/sidoh/esp8266_milight_hub/commit/0b2fbe3f91b5d1b53fb8c5006c15a1f9e1c6d060.  Was a buffer overrun.  Increased the size of the buffer and added a boundary check.  

Will be included in the next release.

Thanks for reporting!

closed 2017-11-11 by sidoh

mcguiresean, 2017-11-12:
Thanks Sidoh!

Love this project - I've tried everything to get a reliable replacement for the milight bridges and nothing has quite worked until this.