sidoh/esp8266_milight_hub #157: MQTT Integration with Home Assistant problems.

mdobossy, 2017-11-13:
I am trying to integrate a single LED strip connected to a limitless LED RGBW Strip controller.  I have everything up and running well, directly from the ESP8266 web interface (and even through the Milight app).  However, when I use the wiki instructions to use the MQTT/MQTT json service through Home Assistant, things get quite flakey.

1. The bulb color in the Hass web interface always stays black, and doesn't show the new color I selected.
2. The actual LED strip color is not the color I selected from the color selector in the Hass web interface (e.g.- I select a shade of green, the strip light turns blue.. or even white sometimes).
3. If I turn the light off (again in the Hass web interface), sometimes it randomly flips back on a few seconds later.

I suspect something isn't quite configured correctly, but am not sure if it is on the ESP hub side, or the Hass side.  I am running esp8266_milight_hub v 1.6rc4, and have the following settings:

On the hub
```
{
  "admin_username": "",
  "admin_password": "",
  "ce_pin": 16,
  "csn_pin": 15,
  "reset_pin": 0,
  "radio_interface_type": "nRF24",
  "packet_repeats": 50,
  "http_repeat_factor": 1,
  "auto_restart_period": 0,
  "mqtt_server": "<REDACTED>:1883",
  "mqtt_username": "homeassistant",
  "mqtt_password": "<REDACTED>",
  "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
  "mqtt_state_topic_pattern": "",
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 10000,
  "mqtt_state_rate_limit": 500,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 3,
  "device_ids": [1],
  "group_state_fields": ["state", "brightness", "color", "mode", "color_temp", "bulb_mode"]
}
```

And the appropriate hass yaml
```
mqtt:

light:
  - platform: insteon_local
  
  - name: "LED Strip"
    command_topic: milight/0x1/rgbw/1

    state_topic: milight/updates/0x1/rgbw/1

    <<: &MILIGHT_PARAMS
      platform: mqtt_json
      rgb: true
      brightness: true
```

Any thoughts on what might be going on here?

sidoh, 2017-11-13:
Try using group 0 instead of group 1.  eg:

```
    command_topic: milight/0x1/rgbw/0
```

Would also recommend using `mqtt_state_topic_pattern` instead of `mqtt_update_topic_pattern`.  Messages published to this topic include all known state (brightness, status, color). They're published as retained messages which means HomeAssistant will be sent the state when it boots.

mdobossy, 2017-11-13:
Getting there.  Switching over to group 0 seems to have fixed the issue with incorrect colors, and flakey connection.  However, Hass still doesn't seem to either be getting the color back from the bulb even after switching over to the mqtt_state_topic_pattern.

I used the following for for the mqtt_state_topic_pattern:

```
milight/state/:device_id/:device_type/:group_id
```

I then updated the state_topic in my hass config to point to that new pattern:

```
    state_topic: milight/state/0x1/rgbw/0
```

I also left the mqtt_update_topic_pattern in the settings.  Should I be getting rid of that field entirely?

sidoh, 2017-11-13:
You can clear `mqtt_update_topic_pattern`.  Won't hurt anything, but no reason for it to send both deltas and full states.

Could you attach a MQTT client to those topics and see what the messages look like?  I usually use `mosquitto_sub`:

```
mosquitto_sub -h my-mqtt-server.com -p 1883 -u user -P hunter2 -t 'milight/#' -v
```

Alternatively, if you look at the state of your device in the REST API it should give the same info:

```
curl -vvv http://milight-hub.local/gateways/0x1/rgbw/0
```

mdobossy, 2017-11-13:
Now this is strange.  I realized that my configuration wasn't be properly loaded.  Once I switched the state_topic on the Hass server over to the mqtt_state_topic_pattern (vs. the update topic pattern), the ESP8266 stops responding to any requests (both through the web interface, and through the rest API).

Even stopping hass, and restarting the ESP8266 has no effect.. the thing won't respond to any requests.  Clearing the flash, and re-flashing the milight-hub firmware got it back up and running again.

sidoh, 2017-11-13:
Huh.  Can't think of a reason that would happen.  If the state topic pattern setting got set to the same thing as the update topic pattern it's possible for it to get into a dumb infinite loop where its own updates trigger commands.  But should probably not cause it to be totally unresponsive.  If it happens again would be helpful to get serial logs.

mdobossy, 2017-11-13:
Ok, tried it again.  I started off with my Hass server stopped.  Re-flashed the NodeMCU, got it on my wifi network.

I went to the web interface on the, and was able to easily play with the lights.  I then setup all of the MQTT fields, and saved the changes.  All was still good.  I could use the REST interface, web interface, etc.

Next, I started up my Hass server.  All was still good.  Went to the light in the Hass web interface, turned it on, and the ESP8266 stopped responding as before.  Every 10 seconds or so, the blue LED on the actual 8266 chip blinks, and about every 10 seconds the blue LED on the NodeMCU board blinks.  But it won't respond either to the milight-hub.local address, nor its IP address.

I grabbed another NodeMCU that I have used for other projects, and the same thing happened.  

I've never tried pulling serial logs from one of these boards before.  If you have a link that describes how one would do so, I'd be happy to get you some logs.

destiny86bln, 2017-11-13:
hey ;)
maybe u should use 

light:
- platform: mqtt_json

instead of

light:
  - platform: insteon_local

sidoh, 2017-11-13:
@destiny86bln -  HASS' YAML structure is unintuitive in places, and I don't think that line is relevant.  The `light` directive takes a list of platforms.  the relevant platform is this, I think:

```yaml
  - name: "LED Strip"
    command_topic: milight/0x1/rgbw/0

    state_topic: milight/updates/0x1/rgbw/0

    <<: &MILIGHT_PARAMS
      platform: mqtt_json
      rgb: true
      brightness: true
```

It would be helpful to attach an MQTT client and see what kind of MQTT traffic is going on.  HASS' event log might be helpful as well.

sidoh, 2017-11-14:
Got it.  So something's causing a crash.  I'll see if I can reproduce.

Are you able to get an MQTT client going?  Would be useful to see what traffic the ESP8266 is getting sent.

Actually that log would only show up if the ESP is getting a remote type (e.g., "rgbw") that it doesn't understand.  It should not cause a crash, but I think it does indicate a problem with the messages the ESP8266 is seeing.

mdobossy, 2017-11-14:
I have an MQTT client going.  Bear with me, as this is my first time doing this.  I set up the MQTT client to connect to the HomeAssistant Broker.  It looks like there are only two possible subscription topics:

```
milight/0x1/rgbw/0
```
and
```
milight/updates/0x1/rgbw/0
```

I don't see any traffic happening with either of them when I try to connect the hub.  If I issue a light on command from Hass, I see a message that looks like:

```
{"state": "ON"}
```

But the problem is- the hub seems to crash before sending any commands from Hass.  Simply enabling the mqtt_state_topic_pattern seems to kill it.

sidoh, 2017-11-14:
That's pretty weird.  I think the next step would be: to compile `MQTT_DEBUG` and `DEBUG_PRINTF` flags enabled and check out the serial logs.  

Could you also paste your most recent settings and the output of `/about` (mostly want to know what version you're running so I can decode the stacktrace).

By the way, you can use a wildcard when subscribing to MQTT topics, e.g. `milight/#` to get anything with that prefix.

mdobossy, 2017-11-14:
Thanks for all the help on this.  I have some good news- everything is working now, and I think the culprit is/was:

The built in Hass mqtt broker.

I decided to install mosquitto, and use it instead of the built in broker, and everything now works perfectly.

sidoh, 2017-11-14:
Interesting.  Glad to hear you got it working.

Would still be curious to see serial logs with MQTT_DEBUG and DEBUG_PRINTF enabled to see if it's a fixable problem.  At the very least it shouldn't be crashing.

Were you using v1.6.0-rc4?

mdobossy, 2017-11-14:
Yes, I am using v1.6.0-rc4.

Below is the serial output from a build with MQTT_DEBUG and DEBUG_PRINTF enabled.  I was able to launch hass without it crashing.  However it crashed as soon as I sent a "on" signal from Hass to the hub.

```
--- Miniterm on /dev/cu.SLAB_USBtoUART  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
MqttClient - Connecting to: 192.168.1.112:1883
MqttClient - connecting
MqttClient - subscribing to topic: milight/+/+/+
MqttClient - Successfully connected to MQTT server
MqttClient - Got message on topic: milight/0x1/rgbw/0
MqttClient - device 0001, group 0
Sending packet (50 repeats): 
B0 00 01 00 00 01 04 
Elapsed: 279
MqttClient - publishing update to milight/updates/0x1/rgbw/0
MqttClient - Got message on topic: milight/updates/0x1/rgbw/0
ERROR - tried to fetch remote config for type
MqttClient - device 0000, group 0

Exception (28):
epc1=0x40210904 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff1c50 end: 3fff2130 offset: 01a0

>>>stack>>>
3fff1df0:  3fff1e90 3ffeaf10 3ffe8374 40223f6a  
3fff1e00:  3fff1e20 3fff1e10 00000004 00000000  
3fff1e10:  00000000 3fff697c 00000000 4021094d  
3fff1e20:  0000000d 0000002a 00000000 3fff1e80  
3fff1e30:  3fff1e90 00000000 3fff3b74 4021acbb  
3fff1e40:  696c696d 00746867 7665643a 5f656369  
3fff1e50:  3a006469 69766564 745f6563 00657079  
3fff1e60:  6f72673a 695f7075 3fff0064 4021ab76  
3fff1e70:  74617473 4e4f0065 4e4f2200 40007d22  
3fff1e80:  3ffeaf40 3fff1e90 00000190 00000018  
3fff1e90:  3fff1e80 3fff1e98 00000000 3fff1e70  
3fff1ea0:  00000002 3fff1e76 00000020 00000100  
3fff1eb0:  00000002 00000000 00000002 00302200  
3fff1ec0:  3fff4b24 00000004 00000004 00000200  
3fff1ed0:  3fff4b27 3fff1f30 3fff39e8 4020694d  
3fff1ee0:  00000000 00000000 3fff1f34 00000008  
3fff1ef0:  3fff4b24 00302305 3fff39e8 40208cbc  
3fff1f00:  00000005 3fff1f40 3fff46ac 0000002e  
3fff1f10:  3fff3f94 00000100 3fff55c4 3fff39e8  
3fff1f20:  00302200 00000100 3fff1f90 3fff5624  
3fff1f30:  00000022 00002200 3fff39e8 00000100  
3fff1f40:  00000003 00000022 3fff55c4 00008003  
3fff1f50:  3fff3f94 3fff48fc 3fff46ac 00000100  
3fff1f60:  00008003 000000f8 ffffffff 3fff39e8  
3fff1f70:  00000000 3fff3f94 3fff5624 40207928  
3fff1f80:  00000008 3fff1f98 ffffffff 6f726701  
3fff1f90:  00000003 657461fc 30310023 30300022  
3fff1fa0:  00000008 00000000 00000001 00000000  
3fff1fb0:  00000000 00000001 00000000 00000008  
3fff1fc0:  00000008 3fff4b24 00000000 00000000  
3fff1fd0:  3fff2350 00000151 00000151 4010020c  
3fff1fe0:  00000000 3ffe8c7f 00000001 40100688  
3fff1ff0:  40248161 00000000 3fff2060 40106c54  
3fff2000:  4025021e 00000821 00000821 40250235  
3fff2010:  3fff3afc 3fff5184 3fff717c 40100688  
3fff2020:  3fff20a0 3fff20b4 0000001a 0000002f  
3fff2030:  00000013 3fff1e40 3fff1e60 00000029  
3fff2040:  0000002f 0000001f 3fff1e80 3fff1e7e  
3fff2050:  3fff1e79 3fff230a 3fff2034 3fff2020  
3fff2060:  00000000 00000000 3fff1e70 00000002  
3fff2070:  00000027 3fff771c 0000002a 3fff20c0  
3fff2080:  3fff20c0 000a5616 3fff771c 4021ace4  
3fff2090:  3fff20c0 00000001 0000001a 4021ddf5  
3fff20a0:  696c696d 00746867 61647075 00736574  
3fff20b0:  00317830 77626772 3f003000 4021dd10  
3fff20c0:  00000401 00000000 3fff58a4 40219d69  
3fff20d0:  3ffe9698 00000000 00001388 3fff10fc  
3fff20e0:  3fffdad0 00000000 3fff3b74 3fff10fc  
3fff20f0:  3fffdad0 00000000 3fff3b74 4021a53c  
3fff2100:  00000000 00000000 3fff10f4 402228e0  
3fff2110:  3fffdad0 00000000 3fff10f4 40202de4  
3fff2120:  feefeffe feefeffe 3fff1110 40204de0  
<<<stack<<<
```

sidoh, 2017-11-15:
Ah, I think I see what's happening.  HASS' MQTT broker isn't following the single-level wildcard to specifications. 

Clipping logs for relevant messages:

```
MqttClient - subscribing to topic: milight/+/+/+
MqttClient - publishing update to milight/updates/0x1/rgbw/0
MqttClient - Got message on topic: milight/updates/0x1/rgbw/0
```

Which means, I think, HASS sent espMH a message on the topic `milight/updates/0x1/rgbw/0` when it was only subscribed to `milight/+/+/+`.  The last `+` shouldn't match `rgbw/0`.

So basically espMH is trying to find a remote type named `0x1` because that's where the `:device_type` token is.  It shouldn't cause a crash, but this is definitely a problem with HASS' MQTT broker.

If it's for some reason better for you to use HASS' MQTT broker, I suspect using `milight/commands/...` as your topic pattern instead of `milight/...` would fix the issue.

sidoh, 2017-11-15:
Thank you for taking the time to provide the logs.  I should be able to fix the crash at least.

mdobossy, 2017-11-15:
No problem, glad it was helpful!  If you need me to test anything else, let me know.

I have no preference on what MQTT broker I use, and thus have mosquitto back up and running again.

sidoh, 2017-11-15:
Crash fixed here: https://github.com/sidoh/esp8266_milight_hub/commit/78284a65b86a5f0c37417ae41bab2fc252269a93.

Thanks again!

closed 2017-11-15 by sidoh