sidoh/esp8266_milight_hub #158: n00b problem with v1.6.0rc4 new unit setup

khmann, 2017-11-14:
I bought this nodeMCUv2: https://www.amazon.com/gp/product/B010O1G1ES/
I tried to flash the file esp8266_milight_hub_nodemcuv2-1.6.0-rc4.bin, several times under OS/X with the git esptool.

Every time, I got an encrypted network shown ESP5431332 which I couldn't connect wether I used password "configesp" or anything else it was the same

I flashed the release v1.5 and the network was open, setup completed, connected to my network.

I then re-flashed the 1.6rc4 again via esptool and everything was still working.  Seems something is strange in the "setup wifi", just fyi.

khmann, 2017-11-14:
update.  Clicking "clear wifi config" button has the same affect.  must reflash v1.5 to get the "connectable ssid" back

khmann, 2017-11-14:
Sorry, I'm a n00b, I found "closed issue #156".  nvm.

closed 2017-11-14 by khmann

sidoh, 2017-11-14:
Haha glad you figured it out.  Definitely pretty confusing while v1.6.0 is still in prerelease.  Planning on releasing soon so hopefully won't be a problem for much longer :)