sidoh/esp8266_milight_hub #159: RF Remote MQTT publishing

jptrsn, 2017-11-15:
In the latest update to 1.6.0, MQTT messages published on RF remote events have changed, breaking some of my functionality. The payload now only contains the command, whereas it previously included the group_id and device_id. I can parse these from the topic itself, but was wondering if I could change a setting in the GUI somewhere to restore these.

I've got an additional question regarding the RF remote, and the group to which a message is published. The master I/O buttons operate differently from the zone I/O buttons. When turning on the master, the group is 0, but when turning off the master, the group is set to the last-used group. Is this a limitation of the RF remote itself (i.e. how the messages are sent from the remote to the hub), or is it a software decision in the MiLight Hub code?

I can understand why this would be the desired function, but it prevents me from sending a { "state":"OFF"} message to zone 0 without first sending a state ON command. I'm using node-red with MQTT to use the remote commands with Home Assistant to control various things dynamically, rather than controlling the lights directly.

Great work, as always!

sidoh, 2017-11-15:
Ah, sorry that broke the workflow.  Yes, my thought was that the information is already contained in the topic, and adding them to the message as well just bloated things.  There's not currently a setting to re-enable it, but could add them to group_state_fields if it'd be valuable. 

espMH should just be decoding the packets as sent by the remotes.  It shouldn't change anything.  Can you tell from sniffed packets if the group IDs are being decoded incorrectly?

jptrsn, 2017-11-15:
Sniffing has revealed that the remote is designed that way. Oh well.

In fact, it turns out that no matter what off button I press, it will send the same zone as the last on button press. Pretty stupid design for a remote, but I guess that's what I got. Should be able to handle it in software by using the on button as a zone selector, and the off button as a toggler. Or something.

I'll just parse out the group ID and remote ID from the topic. Nothing critical was affected, as I'm just getting things set up, and was pretty simple to stitch in upstream of the logic expecting it.

closed 2017-11-15 by jptrsn

sidoh, 2017-11-15:
Noticed the same thing on my remote.  Really is a pretty silly design.

Sounds good.  Glad you got it working.