sidoh/esp8266_milight_hub #160: Black bulbs in HomeAssistant

garmck, 2017-11-17:
Hi, using 1.6.0 and was hoping the bulb color status icon in Home Assistant would work, which it does, but for color only. When bulbs are set to white, they are black, if the bulb color is changed it shows correctly within HomeAssistant and when changed back to white, the icon sticks to the last known color.

My config
  - platform: mqtt_json
    name: "Dining"
    state_topic: "milight/update/0x1167/rgbw/1"
    command_topic: "milight/command/0x1167/rgbw/1"
    brightness: true
    rgb: true
    optimistic: false

Here's the MQTT from turning light on, changing colors and back to white again.
_milight/command/0x1167/rgbw/1 {"state": "ON"}
milight/state/0x1167/rgbw/1 {"state":"ON"}
milight/update/0x1167/rgbw/1 {"state":"ON","bulb_mode":"white"}
milight/command/0x1167/rgbw/1 {"state": "ON", "color": {"r": 255, "g": 40, "b": 9}}
milight/state/0x1167/rgbw/1 {"state":"ON"}
milight/state/0x1167/rgbw/1 {"hue":8}
milight/update/0x1167/rgbw/1 {"state":"ON","color":{"r":255,"g":33,"b":0},"bulb_mode":"color"}
milight/command/0x1167/rgbw/1 {"state": "ON", "color": {"r": 47, "g": 255, "b": 14}}
milight/state/0x1167/rgbw/1 {"state":"ON"}
milight/state/0x1167/rgbw/1 {"hue":112}
milight/update/0x1167/rgbw/1 {"state":"ON","color":{"r":33,"g":255,"b":0},"bulb_mode":"color"}
milight/command/0x1167/rgbw/1 {"state": "ON", "color": {"r": 251, "g": 255, "b": 255}}
milight/state/0x1167/rgbw/1 {"state":"ON"}
milight/state/0x1167/rgbw/1 {"command":"white_mode"}
milight/update/0x1167/rgbw/1 {"state":"ON","bulb_mode":"white"}_

Any ideas?

sidoh, 2017-11-17:
Ah, I misunderstood the issue.  This is definitely the case for me as well, and I agree that it's annoying.

I think HomeAssistant needs to be told what "color" the bulb is in while in white mode.  Even white temperature isn't enough.  I'm not quite sure what the appropriate way to handle that is.  

I actually think this might be a deeper problem with HomeAssistant.  I also have one of these Magic UFO LED controllers which is RGBW.  When in white mode, the bulb color is changed to black.

I'll be happy to poke at this when I get some time, but would be very grateful for any help digging through HomeAssistant documentation/source code.  The [community boards](https://community.home-assistant.io/) might be helpful as well.

sidoh, 2017-11-17:
Code is here, I think.  Haven't taken the time to read it carefully yet:

https://github.com/home-assistant/home-assistant-polymer/blob/91c8b4e/src/components/entity/state-badge.html#L90-L138

garmck, 2017-11-17:
Thanks Chris, black icon not really a deal breaker can get by if its too much issue.

Had a quick look over code (admittedly not a programmer) but looks like the yellow icon is defined in: https://github.com/home-assistant/home-assistant-polymer/blob/91c8b4e/src/components/entity/state-badge.html#L24-L36

_ha-state-icon[data-domain=light][data-state=on]_ - sets to yellow 
_ha-state-icon[data-state=unavailable]_ - sets to black ?

So looks like homeassistant is not reading the correct value for data-state=on ?

garmck, 2017-11-17:
@sidoh i think i've figured it out.
If you add RGB 255,255,255 to the state attributes the icon will display yellow. 
So looks like you just need to add RGB 255,25,255. When sending "bulb_mode:"white" ?
This explains why the bulb stays the same colour as the RGB values are not cleared.
Hope that makes sense?

sidoh, 2017-11-18:
Ah, that makes sense.  Could add a new value to `group_state_fields` -- something like `hass_color` -- which sends (255,255,255) when the device is in white mode.

sidoh, 2017-11-19:
@garmck, could you try the [computed_color](https://github.com/sidoh/esp8266_milight_hub/commits/computed_color) branch?  From my testing with HASS, it seems to have the intended effect.

Note that you'll need to update `group_state_fields`:

[image]

(make sure `computed_color` is selected, and `color` is not selected).

oakbrad, 2017-11-19:
I am seeing the same behavior in Home Assistant. Happy to test a binary for wemos d1, not set up to compile on my current computer

sidoh, 2017-11-19:
Great, thanks! 

This should work for d1 mini or nodemcuv2:

[firmware.bin.zip](https://github.com/sidoh/esp8266_milight_hub/files/1486004/firmware.bin.zip)

oakbrad, 2017-11-19:
Works great for me! All icons as they should be

garmck, 2017-11-19:
@sidoh works great! thanks for sorting and glad major recoding wasn't needed 😄

sidoh, 2017-11-20:
Sweet.  Thanks for testing guys.  Think I'm going to make this the default (instead of `color`).  I'll make a patch release shortly.

sidoh, 2017-11-20:
Fixed in [1.6.1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.6.1).

Thanks @garmck for reporting and helping track down the issue!

closed 2017-11-20 by sidoh