sidoh/esp8266_milight_hub #162: request: mqtt to http

kurniawan77, 2017-11-20:
Hi,

i guess this isn't an issue but more of a request.

I am running ESPMH 1.6.0 atm with Home Assistant and [diyHue from Marius Motea](https://github.com/mariusmotea/diyHue) to control my MiLights. This works great and i just started testing it with mqtt and Home Assistant. What i discovered was that this works and updates way faster than Hue (HTML) in Home assistant. But i also love the hue app for controlling the lamps only on my mobile. So i then came up the idea to use mqtt in Home assistant and the Hue app on mobile. Right after one test i discovered that if controlled through mqtt, the hue app doesn't get the status update but controlled from hue app (html), mqtt gets updated. 

Would it be possible for ESPMH to act as a Hue/Mqtt bridge and send it's status to the diyhue bridge?

kind regards

sidoh, 2017-11-21:
Hi there,

I think I'm not exactly following how all of the pieces connect.  ESPMH can be configured to send the state of devices upon a change to a customizable MQTT topic.  Would that accomplish what you're hoping for?

kurniawan77, 2017-11-21:
Hi,

I don't think so. I guess it should be the other way around... can ESPMH send the state of device upon change to Hue protocol? 

I control the Milights as a Hue light with [diyHUE](https://github.com/mariusmotea/diyHue/wiki/4.-MiLight-Hub)

sidoh, 2017-11-21:
I think probably the best approach is to somehow glue ESPMH mqtt_json devices to your diyHue devices using HomeAssistant.  I'm not familiar enough with how diyHue works to suggest anything in much detail, but I'd imagine it's possible.  Perhaps an automation that forwards stuff from diyHue devices to the mqtt_json devices on state change?

kurniawan77, 2017-11-21:
Thanks for your reply. 

i think it's maybe best to ask Marius. It works fine one way (switch on light with hue app > milights goes on, also in mqtt).  But the other way around it doesnt update the light in hue app (switch off with milight remote or mqtt > milights goes off but in the hue app status stays on).

That's why in first i thought to ask here as because it seems that espmh wasn't sending it status to hue api

Kind regards

sidoh, 2017-11-21:
The state for a device is sent by espMH when:

1. A command to change the state is received.
2. A 2.4 GHz packet from a remote or other milight device is intercepted.

It sounds like diyHue is not configured to react to that update?

kurniawan77, 2017-11-23:
https://github.com/mariusmotea/diyHue/issues/100

kurniawan77, 2017-12-04:
Kinda works now, see link above. The only thing missing is RGB to xy color conversion. Maybe someone can supply some help?

janLo, 2017-12-04:
@kurniawan77, you can try the [emulated HUE bridge](https://home-assistant.io/components/emulated_hue/) from HomeAssistant to control your Homeassistant with a HUE-capable device.

kurniawan77, 2017-12-04:
But i dont have a HUE device... it's all diyHue

janLo, 2017-12-04:
Thats why I said "HUE-capable device". The emulated hue bridge from Home assistant emulates a Hue-Bridge - so you should be able to use all Remotes/Services/Apps/Whatever with it that work with a HUE-Bridge. Just directly from HomeAssistant without the need for diyHUE.

kurniawan77, 2017-12-04:
i don't believe that the emulated hue bridge from home assistant can do what diyHue does.... No way

janLo, 2017-12-04:
It exposes the Home-Assistant connected devices via the HUE api/protocol. What do you think is missing?

kurniawan77, 2017-12-04:
diyHue is able to control ZigBee lights (using Raspbee module or original Hue Bridge or IKEA Tradfri Gateway), Mi-Light bulbs (using MiLight Hub), Neopixel strips (WS2812B and SK6812) and any cheep ESP8266 based bulb from market by replacing firmware with custom one.

janLo, 2017-12-04:
* Zigbee: https://home-assistant.io/components/zigbee/
* MiLight: https://home-assistant.io/components/light.limitlessled/ (or mqtt-light if you use the esp-milight-hub)
* Raspbee: Benutzbar via REST
* (Original) HUE Bridge: https://home-assistant.io/components/light.hue/
* Ikea: https://home-assistant.io/components/tradfri/
* The "cheap-ESP-based bulbs": You probably mean the ones where you flash code from https://github.com/mariusmotea/diyHue/tree/master/Lights to. They just communicate over wifi using a rest-like http-api, That should be easily addable to HA via the REST-Component or a custom one. The Code seems pretty simple: https://github.com/mariusmotea/diyHue/blob/master/BridgeEmulator/HueEmulator.py#L421-L428

kurniawan77, 2017-12-04:
True! You can make all components work in Home Assistant. You can also buy expensive Philips Hue. I chose to **make** my "Philips Hue" because i found the light too expensive. I fell in love with the Hue app and prefer to use the Hue app to control my lights on the go, it has a nice big ui, and let the automation stuff done in Home Assistant. But still, i don't believe what u suggest fits my needs (can HA hue emulator  use the hue app?). With all respect! It's also not my question. 
Like you suggest above in Milight: mqtt light with ESPMH. This is what i use for milights in Home Assistant! Point is/was: diyHue wasn't reacting to status changes, it does now after fix! Well not completely.... diyHue doesn't react to color changes... yet. So my question is maybe someone could help marius with the conversion formula.

janLo, 2017-12-04:
The Emulated HUE from HomeAddidtant provides exactly the same Interface (The HUE API7Protocol) that diyHUE provides and emulates a HUE bridge that exposes all the devices HomeAssistant knows about. The HUE app just uses this protocol. So it should be easily be able to communicate with HomeAssistant.

kurniawan77, 2017-12-04:
do you happen to know how to push the button in Home Assistant?

sidoh, 2017-12-04:
I realize this isn't your question, but I'll echo the suggestion to use HomeAssistant as your HA glue.  If you're using diyHue in this way, chances are HomeAssistant will work just fine, and probably be capable of way more.

As far as converting between the RGB and XY color spaces, there's a ton of prior art out there.  For example, [this javascript library](https://github.com/Q42Philips/hue-color-converter).  If there isn't already a C++ implementation, it'd be easy to port.  Could have espMH support it, but it feels a little superfluous.

kurniawan77, 2017-12-05:
Thanks for both of your advice. I have tried it before and i have tried it yesterday again. Home Assistant emulator just doesn't work with the Hue app, there is no "auto button press to connect". 

Thank you for showing the javascript library, very kind. Closing this issue because it isn't an issue anymore after marius made a fix.

Sorry for complaining here...

Kind regards.

closed 2017-12-05 by kurniawan77