sidoh/esp8266_milight_hub #163: Access Disco Modes as Effects in Home Assistant?

oakbrad, 2017-11-21:
I noticed with the recent update that the disco "modes" are individually selectable and can be called via MQTT/REST. Awesome! Is it possible to integrate these as "effects" as supported in the Home Assistant mqtt_json component?

This would allow them to be selected via the native UI and for Hass to track the mode state.

Thanks for your work on this project!

sidoh, 2017-11-21:
Would be pretty straightforward, I think.  But I can't think of a way to make it pretty without hardcoding assumptions (e.g., "effect 0 = White Flash").  Most straightforward way would be to:

1. Add support for setting mode from `effect` directive [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L381).
2. Add `effect: true` and `effect_list: ...` to mqtt_json config in HASS.

Also, testing this out made me realize that night_mode is not working for RGB+CCT bulbs.

garmck, 2017-11-21:
@sidoh night_mode is working on my FUT035 LED strip which I've paired as RGB_CCT, night mode also working fine for my FUT016 RGBW bulbs.

oakbrad, 2017-11-21:
I was looking at building one of these, and he has supported effects and the speed via mqtt_json if that's helpful. https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs

Speed control would be nice as well as some of the effects are a bit intense on the default fast setting :).

I was trying the effects out for MQTT commands and this were what I labeled them if this is helpful...
0 = rainbow disco
1 = white glow
2 = RGB glow
3 = color cycle
4 = RGB cycle
5 = red flash
6 = green flash
7 = blue flash
8 = all effects loop

sidoh, 2017-11-21:
Yeah night mode works fine on my RGBW bulbs as well.  It just turns off my FUT105s, though.  I'll make a different issue for it.

Speed is a little tricky because no device supports a way to directly set it.  There are only increase and decrease commands.  It's possible to hack around this, but it's pretty gross, and requires a lot of packets.

Thanks for labeling the effects.  The trouble is that the effects are different depending on the device.  For example, mode 0 for my FUT105s is an orange flicker (something like a shitty fire lol).  It'd be nice if HomeAssistant had a way to label the effects separately from the value that gets sent in the command packet.  Not sure if that's a possibility.

sidoh, 2017-11-23:
Think this branch should do the trick:

https://github.com/sidoh/esp8266_milight_hub/tree/support_hass_effects

You'll need to update your HASS config to something like this:

```
- name: "Lamp"
  command_topic: milight/0x1/rgb_cct/1
  state_topic: milight/states/0x1/rgb_cct/1
  <<: &MILIGHT_PARAMS
    platform: mqtt_json
    color_temp: true
    rgb: true
    brightness: true
    effect: true
    effect_list:
      - white_mode
      - night_mode
      - 0
      - 1
```

and make sure that the "effect" field is enabled in the espMH config:

[image]

Can't think of a nice way to make the effect names better without a ton of hardcoded maps.  Wish `effect_list` in HASS was a dictionary.

sidoh, 2018-03-12:
Forgot to close the issue after adding support in 1.6.

closed 2018-03-12 by sidoh