sidoh/esp8266_milight_hub #164: Support for smooth on/off

bhaap, 2017-11-21:
As discussed in this ticket:

https://github.com/sidoh/esp8266_milight_hub/issues/113#issuecomment-345977067

bhaap, 2018-03-01:
Nothing on this?

bhaap, 2018-03-01:
> As a easy feature to add would be to make the OFF command send brightness to 0 wait for X ms depending on current brightness and then send OFF.

And ON command send ON then Brightness 255 if no other brightness was given.

Easy way to get a smooth on/off atleast.

The only downside to transitions by the "hub" is that we get alot of radiotraffic, say swapping from one color to another. That would take like atleast 100 radio commands to get a smooth transition.

This would be alot easier if the bulb had built in transitions.

sidoh, 2018-03-01:
@bhaap, I'd be happy to look at this when I've decided to set aside focused time to work on this project.  As of now, I'm preoccupied with other stuff.  My work on these kinds of things comes in bursts :)

I'd also be more than happy to collaborate on a PR if you're interested in taking a stab.

bhaap, 2018-03-01:
Only a novice in programming but ofcourse i can give it a stab. Could you give me some pointers on where to look in the code for sending the on and the off command (guessing its a specific place depending on bulb?)

poudenes, 2019-05-31:
Is this still in progress? Sounds nice.. Smooth on and off.. Then my apartment will look like a mansion for outsiders :)

sidoh, 2019-09-01:
This is now possible with transitions (documented in README).

closed 2019-09-01 by sidoh