sidoh/esp8266_milight_hub #165: Phantom MQTT Commands

jptrsn, 2017-11-22:
I updated Home Assistant (HassIO) to 0.57.3, and the hub to 1.6.1. For some reason, the interaction of the two results in a command being issued to my device 0x2 (a 9w milight rgbw bulb) to enter mode 5. Looking at the MQTT command topic for this bulb that's defined in Home Assistant, there isn't a command being issued. Sniffing the 2.4 GHz packets, there are no commands being issued to this device (which I wouldn't expect, since it's an RGBW light and I don't have an RGBW remote).

Looking at MQTT, I'm seeing events changing the mode of that bulb, but not of any other bulbs. I've shut down Home Assistant and left Mosquitto running, and I'm still seeing the command being issued. I've checked all the devices on my network to see if any have an open session, but nothing. When I restart the ESP, it appears to issue the command to set mode to 5. I see the state of that device get updated, and the updates topic shows only {"mode":5}

Is there any way to "factory reset" the firmware? Do I need to rebuild and upload? What could be causing this to happen? I suspect it's related to the states feature that was recently added. I can't wrap my head around why it's this particular combination - I've tried setting the mode to a different integer, which issues the command successfully, but the bug is always setting mode 5.

sidoh, 2017-11-23:
I agree this sounds related to the state changes.  Could you paste some logs from MQTT?  Would be helpful to know the full topics and messages.  I usually use `mosquitto_sub` like this:

```
$ mosquitto_sub -h 10.133.8.11 -p 1883 -u user -P hunter2 -t 'milight/#' -v
```

jptrsn, 2017-11-23:
I used the web UI to set the bulb to my preferred state of on at full brightness, then subscribed to topic, update, and state. Here's the settings and the resultant output.

Hub is configured as follows:
**mqtt_topic_pattern** : `milight/:device_id/:device_type/:group_id`
**mqtt_update_pattern** : `milight/updates/:device_id/:device_type/:group_id`
**mqtt_state_topic_pattern** : `milight/states/:device_id/:device_type/:group_id`

I put the light into "white" mode by opening the hub ip address in Chrome, then clicking the "White" button. Bulb state was on, full brightness, white.

*Output when first subscribing to topics:*
**MQTT Command Topic**
`mosquitto_sub -h [redacted] -u "[redacted]" -P "[redacted]" -t "milight/0x2/#" -v`
`milight/0x2/rgbw/1 {"state":"ON","mode":5,"bulb_mode":"scene"}`

**MQTT Update Topic**
`mosquitto_sub -h [redacted] -u "[redacted]" -P "[redacted]" -t "milight/updates/0x2/#" -v`
*No initial output. After a short delay...*
`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/updates/0x2/rgbw/1 {"mode":5}`

**MQTT State Topic**
`mosquitto_sub -h [redacted] -u "[redacted]" -P "[redacted]" -t "milight/states/0x2/#" -v`
`milight/states/0x2/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"white"}`
*After a short delay...*
`milight/states/0x2/rgbw/1 {"state":"ON","brightness"255,"mode":5,"bulb_mode":"scene"}`

I then set the light to it's preferred state, on full brightness white, in the Home Assistant UI. Here's what comes up in those topics:
**Command**
`milight/0x2/rgbw/1 {"state": "ON", "color": {"r": 255, "g", 253, "b": 251}}`
**Update**
`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/updates/0x2/rgbw/1 {"command":"white_mode"}`
**State**
`milight/states/0x2/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"white"}`

Again, a short while later, I see this happen, but curiously, *there's no change to the light itself*
**Command**
`nothing new`
**Update**
`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/updates/0x2/rgbw/1 {"mode":5}`
**State**
`milight/states/0x2/rgbw/1 {"state":"ON","brightness":255,"mode":5,"bulb_mode":"scene"}`

A short while after that, I see the same messages again, and this time, the bulb changes state to mode 5. If it would be helpful, I can give you the full output of all messages from all sub-topics of milight/# (although if anyone can tell me how to copy from a putty session, that would be great, because I've retyped all this content manually since CTRL-C stops the process, and right-click duplicates the highlighted text).

sidoh, 2017-11-23:
Thanks for providing all of this information.

This is quite weird, especially since I can't seem to reproduce it.  This is what I see for my RGBW bulbs (just clicking around in the UI):

```
milight/updates/0x1/rgbw/1 {"state":"OFF"}
milight/states/0x1/rgbw/1 {"state":"OFF","brightness":255,"bulb_mode":"color","color":{"r":0,"g":255,"b":67}}
milight/updates/0x1/rgbw/1 {"state":"ON"}
milight/states/0x1/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"color","color":{"r":0,"g":255,"b":67}}
milight/updates/0x1/rgbw/1 {"hue":64}
milight/states/0x1/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"color","color":{"r":238,"g":255,"b":0}}
milight/updates/0x1/rgbw/1 {"mode":4}
milight/states/0x1/rgbw/1 {"state":"ON","mode":4,"bulb_mode":"scene","color":{"r":255,"g":255,"b":255}}
milight/updates/0x1/rgbw/1 {"hue":321}
milight/states/0x1/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"color","color":{"r":255,"g":0,"b":170}}
milight/updates/0x1/rgbw/1 {"command":"white_mode"}
milight/states/0x1/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/updates/0x1/rgbw/1 {"brightness":122}
milight/states/0x1/rgbw/1 {"state":"ON","brightness":122,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
```

Because messages published to `mqtt_update_topic_pattern` are decoded from raw milight packets, -- either sent locally or intercepted from another device -- I'm quite sure that the presence of this:

```
milight/updates/0x2/rgbw/1 {"mode":5}
```

means that either:

1. espMH is receiving a 2.4 GHz packet that's setting the mode to 5, or being instructed in some other way (REST...somehow?) to set the mode to 5.
2. something else is publishing that message 

You don't happen to have another install of espMH would you?  Or anything else that could publish to these MQTT topics?  Is HASS configured to do anything special with the milight MQTT topics?

It would also be helpful to get serial logs with `DEBUG_PRINTF` and `MQTT_DEBUG` enabled if at all possible.  I can provide a binary if that's helpful.

sidoh, 2017-11-23:
(and been a while since I've used putty, but i think it at least used to be the case that selecting text copied it to the clipboard)

jptrsn, 2017-11-23:
I fried my USB-to-serial a couple weeks ago, and I'm waiting on delivery of another. If you can provide a binary for an ESP07 that I can upload OTA, that would be super helpful. However, if it's gonna need a serial converter, I'm out of luck for the time being.

I've got no other hubs or devices locally. I've disabled all automations, searched the config directory for any instance of `5` or `mode` and nothing comes up, and I can't imagine how anything would be hitting the REST API unless I've had a serious network security breach.

I've unplugged the hub, and left the subscribe commands running in putty. If the source is an external MQTT command, they should appear in the MQTT topics despite the hub being disconnected. If it's external 2.4 GHz, then I won't see anything until I reconnect the hub. I'll leave it disconnected overnight, then power it back on tomorrow morning and sniff the RF traffic to see if it's an external RF packet triggering things. I didn't see any today, but I wasn't taking a disciplined approach to narrowing in on the issue.

Odd that it's only affecting the 0x2 device. My other two bulbs are the 6w RGBW variety, and those are working fine. If I can rule out all external influence, then my next step will be to pair the bulb to a different ID and see if anything changes. I've chosen these IDs (0x1, 0x2, and 0x3) completely arbitrarily, so I can easily change to something a bit less obvious, and see if it's only affecting the device ID or the hardware attached to the ID.

sidoh, 2017-11-23:
Actually I think we'd get most of the benefit from the "Sniff Traffic" feature in the UI.  If you enable that while you're running your tests we should be able to see what RF packets espMH is seeing/sending, but we won't be able to tell the difference between sent and received.

Yes, definitely weird this is only affecting one device.  Do you see the same behavior when using the UI?

If you could provide:

1. Full MQTT logs (from `milight/#`).  Will be useful to know the order in which messages are being received.
2. Output of "Sniff Traffic" from the UI.

we might get some good clues.

jptrsn, 2017-11-23:
Overnight, I sniffed the 2.4 GHz traffic, and there was absolutely nothing there. No commands appeared at all, which struck me as odd. I thought I'd at least see the commands issued by the espMH, but there were none.

I started with the bulb in white mode, turned off. I then subscribed to MQTT, and turned the light on. Here's the full output from all milight topics.

`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/states/0x2/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"white"}`
*After these commands, the bulb is in the on state, at full brightness, white.*

`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/updates/0x2/rgbw/1 {"mode":5}`
`milight/states/0x2/rgbw/1 {"state":"ON","brightness":255,"mode":5,"bulb_mode":"scene"}`
*After these commands, the bulb is still in the same state, full brightness white, turned ON.*

`milight/updates/0x2/rgbw/1 {"state":"ON"}`
`milight/updates/0x2/rgbw/1 {"mode":5}`
*Now the bulb is in mode 5, flashing red.*

Again, no RF traffic was sniffed whatsoever. I do not see anything sniffed when I issue commands, either through the web UI or via Home Assistant. It looks like any RF traffic initiated by the espMH doesn't appear in the log.

sidoh, 2017-11-23:
I'm testing with 1.6.1, and stuff sent by espMH is definitely showing up in sniffed traffic for me.  example:

[image]

Things are getting sent over by a websocket.  Perhaps the connection got interrupted.  Maybe try refreshing and re-enabling sniffing?  If that's still not working, maybe check browser devtools.

Are you using a pre-compiled binary, or did you compile the firmware yourself?

jptrsn, 2017-11-23:
I'm using the precompiled binary for the ESP07. You're right, the commands do appear in the devtools console. I can see there's an update to 1.6.2 available. Do you recommend installing, or should I hold off so I can continue to troubleshoot?

In the console, I can see this object get logged as soon as I select 0x2 from the drop-down menu:
`{brightness
:
255
bulb_mode
:
"scene"
mode
:
5
state
:
"ON"}`

I had thought that there might be some issue with the bulb itself, so I disconnected the bulb (fully removed from the socket), but still see the MQTT updates getting posted.

When I set the bulb to white in the web UI, I can see 
`{brightness
:
255
bulb_mode
:
"white"
state
:
"ON"}` logged twice in the console, both from line 3 of index (although inspecting, it looks like all the code is on a single line).

jptrsn, 2017-11-23:
Checking the settings, I had auto_restart_period set to 5, which roughly corresponds to the problem. It's happening on any power cycle, so I'm assuming that it's a state that gets assigned at boot. I reset it to 0, refreshed the page, and now I'm seeing the packets in the sniffer. I'll keep the window open and see if there's any difference (I suspect there won't be any more phantom commands without the device auto-rebooting).

sidoh, 2017-11-23:
Good catch!

1.6.2 just has some fiddly changes to better handle night mode.  Updating shouldn't make a difference.

I wonder if the mode command `{"mode":5}` got published as a retained message.  When you first connect with `mosquitto_sub`, do you see a `{"mode":5}` message?  You can clear a retained message by submitting an empty packet with the retain flag, e.g.:

```
mosquitto_pub -h 10.133.8.11 -u user -P hunter2 -t 'milight/states2/0x1/rgbw/0' -m '' -r
```

jptrsn, 2017-11-24:
That was definitely it - a retained message. I'm not sure how it got set, and I'm not clear on how to reset all retained messages in Mosquitto running on HassIO, but publishing a null message with the retain flag has fixed the issue.

Thanks for sorting through this with me. Was an interesting problem-solving process. Now to get playing with night mode. I'm using node-red to manage automations and scenes, so should be able to get some interesting behaviours out of these bulbs.

Thanks again for the great work on this project!

closed 2017-11-24 by jptrsn

sidoh, 2017-11-24:
Glad you've got it figured out.

espMH publishes state messages with the retained flag.  Could imagine this might've happened if `mqtt_state_topic_pattern` was set to the same thing as `mqtt_topic_pattern` at some point.

sidoh, 2017-11-24:
HASS' `mqtt_json` component also has a `retain` option that might've caused this:

https://home-assistant.io/components/light.mqtt_json/