sidoh/esp8266_milight_hub #167: Prevent bulbs to go into "white mode"

janLo, 2017-11-23:
I have a milight controller for an RGB led-strip. I use it from HomeAssistant via MQTT as rgbw device. Sadly it goes to "white mode" when the color gets too "bright". This causes the device to turn off all rgb channels and turn on the "white channel" which is not connected.

Is there a possibility to prevent this or can this be implemented as a new bulb type? Just using "rgb" instead of "rgbw" causes the light to flash strangely.

janLo, 2017-11-23:
MQTT dump:
```
milight/0x1F25/rgbw/1 {"color": {"r": 255, "b": 255, "g": 255}, "state": "ON"}
milight/updates/0x1F25/rgbw/1 {"state":"ON"}
milight/updates/0x1F25/rgbw/1 {"command":"white_mode"}
milight/state/0x1F25/rgbw/1 {"state":"ON","brightness":82,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
```

janLo, 2017-11-23:
There is actually a low-tech solution, that is a bit of a hack. Proper SW-Support would be better:
[image]

sidoh, 2017-11-23:
Haha, love your solution with the diodes.  That's great :)

This is done intentionally because it's how HASS users are accustomed to re-entering white mode.  Code is here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L300-L302

Could make this optional behavior.

Your suggestion might make the most sense, though -- implement a new remote type that behaves the same as FUT096 (RGBW) except white mode is disabled.

janLo, 2017-11-23:
I think, the hue value would not matter, the saturation should be zero in the case of white. Can this be set to the lamp?