sidoh/esp8266_milight_hub #168: Color altered if more than one values are the same

janLo, 2017-11-23:
If I set a color where at least two of the color vales (r, g, b) are the same, another color is set and reported back. For example when I set (155,155,155) I get (255,0,0):
```
milight/0x1F25/rgbw/1 {"color": {"r": 155, "b": 155, "g": 155}, "state": "ON"}
milight/updates/0x1F25/rgbw/1 {"state":"ON"}
milight/updates/0x1F25/rgbw/1 {"hue":0}
milight/state/0x1F25/rgbw/1 {"state":"ON","brightness":255,"bulb_mode":"color","color":{"r":255,"g":0,"b":0}}
```
If I change it to (154,155,155) I get (0, 250,255) and so on.

I guess this has to do with the rgb->hue conversion. It just came up with debugging the problem in #167  and it prevents me to find any value that could be described as "emulated white".

sidoh, 2017-11-23:
Ah, yep.  That does sound like an issue with the RGB -> hue conversion.  That's happening here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L300-L312

I'm not exactly sure what the correct behavior is given that color is controlled by hue.  I don't think there's a value for hue (assuming full saturation) that resembles white.

janLo, 2017-11-23:
I think, the hue value would not matter, the saturation should be zero in the case of white. Can this be set to the lamp?

sidoh, 2017-11-23:
It's unclear to me how a grayscale color should be represented.  Should it be white mode with appropriate brightness?

Setting the saturation to 0 sets the white LED brightness to max, but the RGB LEDs are still on, so there's a very obvious color.

The knobs that the actual devices have are:

1. Hue: 255 predefined mixes of R, G, and B brightnesses
2. Saturation: 100 predefined brightness values for the white LEDs.  0 is max brightness, 100 is off.
3. Brightness: 100 predefined brightness scaling factors applied universally to all LEDs (R, G, B, and W).

The most sensible way to interpret a color like RGB=(x,x,x) is, I think, to turn the RGB bulbs off and scale the brightness down to (x/255)*100.  But since brightness is controlled independently, I'm not sure it makes a ton of sense to also control in this way.

There are real color spaces that behave this way.  For example, [CIE 1931](https://en.wikipedia.org/wiki/CIE_1931_color_space).  White and grey are the same color with different brightness values.

I'm very happy to take suggestions on the best way to handle this, but it's still unclear to me how it should behave.

oakbrad, 2017-11-24:
> Should it be white mode with appropriate brightness?

In RGB, yes that is how I would expect it to behave.

It is common to use the average value of the RGB channels as an expression of brightness in image applications.

sidoh, 2017-11-24:
The issue is that brightness is already a separate control.  How should state be represented?  Should setting rgb=(x,x,x) return color:(x,x,x), brightness: 255, or color: (255,255,255), brightness: x?  Either is valid.

My hesitation is that it feels that there's inherent ambiguity in handling grayscale colors, and I'd rather just have folks use brightness as long as that's not going to cause problems.

I agree that current behavior is very unexpected. Perhaps specifying any grayscale RGB color should put the bulb in white mode?

oakbrad, 2017-11-24:
I see what you're saying. This makes a lot more sense to me

> color: (255,255,255), brightness: x

I would expect that  for any value of 'gray' (another way of saying that is 'neutral' since there's no 'gray' light) the bulb would set to white_mode, brightness x

This would be consistent with the Hass UI as well, right? When you select a brightness for a light, you're not selecting RGB gray, you're changing the brightness of white light.

sidoh, 2017-11-24:
Okay, fair enough.  It just feels weird to have any command that isn't perfectly reflected in the resulting state, but probably makes the most sense here.

Yes, this should work just fine with HASS.  I think HASS always sends 255,255,255.

sidoh, 2019-05-20:
I think the original issue has since been fixed, and unless someone complains, I'm not going to bother with supporting setting brightness via grayscale colors.

closed 2019-05-20 by sidoh