sidoh/esp8266_milight_hub #169: next_mode and previous_mode

servim, 2017-11-25:
Hello! I'm not sure I can explain, my English is quite bad.
I have a light bulb RGBW, firmware 1.6.0
Command: next_mode it works, the mode changes from 0 to 8. But if the mode is changed through the mode, as I showed above: {“state”:“ON”,“mode”:"’+Milight_ID1_G1_mode.state+’"}, then this command will continue from the place where it finished, and not from the last mode. Eg:
[image]
Command: previous_mode, changes mode 15,2,1,0,15,2,1,0 etc
[image]

Is this expected behavior?

sidoh, 2017-11-25:
Ah, yes.  This is not how it should work, but I see why it's doing this.  Should be able to fix this in a future version.  Thanks for reporting.

servim, 2017-11-25:
the question is not the topic:
Command: mode_speed_up and mode_speed_down, works, but it is not clear, the current state, what is the range from the minimum to the maximum? mosquito does not return status.

sidoh, 2017-11-25:
There isn't a way to directly set speed.  It can only be controlled with increase/decrease commands.  Because of this it's tricky to know what speed the bulb is at.  espMH doesn't currently track speed, but I believe the range is [0, 10].

It's possible to know what the state is by, for example, sending the maximum number of decrease commands.  Then you know the device is at speed 0.

servim, 2017-11-25:
Thank you! I thought so :(

sidoh, 2018-07-29:
was added in 1.7

closed 2018-07-29 by sidoh