sidoh/esp8266_milight_hub #170: Sporadic buffer overflow in CCT brightness commands

sidoh, 2017-11-25:
Discussion and detail in this comment thread:

http://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/comment-page-3/#comment-2659

vantees, 2017-12-01:
Hi. I can confirm that with CCT bulbs interface stops working. After sending just one command ON/OFF it does not respond any longer. You have to restart ESP. Tweaking ESP does not help. I have D1_mini and 1.6.2 (d1_mini) firmware.

Skinah, 2017-12-08:
I have the same issue with brightness and colour changing CCT globes. The behaviour is like the espmilight hub is trying to do a transistion as you see the level slowly fade instead of jumping directly to what you select. On and Off is also very very slow to respond. Also the colour temp does not get published to the mqtt state.

sidoh, 2017-12-08:
> The behaviour is like the espmilight hub is trying to do a transistion as you see the level slowly fade instead of jumping directly to what you select. On

This is an unavoidable limitation of the protocol the bulbs use.  With the CCT protocol, there is no command to set brightness or color temperature directly.  There are only increase/decrease commands.

espMH tries to expose a way to set each of these directly with some hackery.  It sends the down command 10 times, followed by the appropriate number of up commands.  This is really brittle and honestly I would not recommend using it in its current state.

A lot of newer CCT devices support some of the newer protocols (like RGB+CCT).  If that's an option for you, would definitely recommend it.

Otherwise, happy to take suggestions on how to better handle brightness and temperature on the old CCT protocol.

khmann, 2017-12-19:
It's a little hard to justify the complexity (underlying code change), but here's how you make it the best:

Right now you send 10x 30 repeats (ie: unique sequence numbers) warm/cold packets to get to a known state.  This is a little choppy and could be inaccurate.  You can save time and get better guarantee by sending 256x unique sequence (don't repeat) warm/cold packets It's enough to ensure the bulb gets where you want it. Save a little time by ramping full warm/cold whichever is closest to target temperature.  Then adjust as normal.

Also, don't those bulbs have a "full on"?  what if you always ramp the bulb down when asked to turn it off via MQTT.  Then when turning on, either "resume" (=lowest) or full on and adjust from there.

The same (always use unique sequence) applies to old RGB protocol brightness as well

sidoh, 2018-01-03:
Great idea @khmann.  I agree that it might get a little hairy to do it this way, but perhaps worth looking into.

@hastarin mentioned the same thing about a "full on" command.  Is there a similar command for white temperature?  That'd certainly be preferable.

There are a couple of places where I'd like the command handlers to be more state-aware, and this is one of them.  Maybe a little more risky to assume state is correct in this case, though.

I won't have time to dig into this for another week or two, but happy to look at a PR if anyone else wants to take a crack.

hastarin, 2018-01-05:
I have found that doubling the buffer (to 280 from 140) prevents the crash.   That does not fix the issue of weird ramps that are enough, at least after playing with them for an hour, to make you ill.

The limitless led plugin for home-assistant resets all the lights on startup to a known state and then assumes that state.  This works reasonably well but still gets out of sync over time.

@sidoh There is no command for white temperature that I'm aware of.

My suggestion, which I'm unsure I have the skills to code, is to apply logic along the lines of the following:

1. Whenever setting brightness to full send the command for full brightness.
2. Similarly whenever setting min brightness, or min/max white temp send some extra packets (say 5).
3. Otherwise assume the state recorded for the lights is correct.

That provides the user with a way to get lights back in sync (by setting min/max levels) and prevents the need to ramp down/up (or up/down).

southwalesboy, 2018-02-16:
I find that everything works great CCT, can send {"state": "ON"} and "off" via MQTT to the hub indefinitely, but if brightness is set at 254 or 255 then the bulb dims and the hub stops responding to CCT bulb commands.

{"state": "ON", "brightness": 253} works fine

{"state": "ON", "brightness": 254} breaks CCT command response

{"state": "ON", "brightness": 255} breaks CCT command response 

Hope this information is useful

Skinah, 2018-02-16:
Correct I posted this a few days ago....
https://github.com/sidoh/esp8266_milight_hub/issues/214
I have written an Openhab2 binding for this project and I found the same thing and by using a workaround the fading works very well and so does colour temp changing, if you workaround the bug/s. I believe it should be very easy to work out the root cause and solve this issue, I'm just too busy with my binding at the moment to look myself as the work around has solved the need to look at it.

sidoh, 2018-03-12:
Going to take a look at this for the 1.7 release.

sidoh, 2018-03-12:
this was a really stupid bug.  da79a43b4a2e76f9e50bb621f05daeb8acd899f0 should fix.

closed 2018-05-05 by sidoh