sidoh/esp8266_milight_hub #171: color -> night -> white

servim, 2017-11-26:
1.6.2 RGBW
If you select a color, then the night mode, and then white, the bulbs will light up with the last color, not white, but in milight/states/XXXX/0 will return {"state":"ON","level":100,"bulb_mode":"white"}

sidoh, 2017-11-26:
I see that bulbs start up with the color, but are quickly switched to white.  This is what I'd expect.  The bulbs will always have to be turned on before changing their state.

Is this not consistent with what you're seeing?

servim, 2017-11-27:
No, I do not see what I expected. I pressed the button - "White", the bulbs light up in color, but in the MQTT status returned - white. Example:
I press the buttons: green - night - white
light bulbs: green - night - green
MQTT:
[image]
and the light is green.

sidoh, 2017-11-30:
Ah interesting.  That should definitely not be happening. 

I think I see what's happening.  Before I was testing with HASS, which always sends `{"status":"ON"}` along with the command.  So it was sending `{"effect":"white_mode","status":"ON"}`.  This has the expected behavior.

The bulbs basically behave like they're off when in night mode.  They only respond to ON and OFF.

Debating what makes most sense here.  Think it probably makes sense to force bulbs into ON when a command is received in night_mode.  What do you think?

servim, 2017-12-04:
Speaking frankly, I do not really understand English)))
It's hard for me to understand what you are asking, I'm sorry.
I tried to show you pictures where I think there is a problem. I hope you got what I meant ;)

sidoh, 2017-12-04:
No worries.  I understand the issue, and should be able to fix it in a future release.

fab33, 2017-12-28:
Doesn't know if it's related but for a rgbw bulb, if you publish ```{"state":"ON","level":100,"bulb_mode":"white"}```
status is not updated and bulb doens't change to white. (ESPMH 1.6.3)

sidoh, 2017-12-28:
I think `bulb_mode` is currently a read-only state field, which is kind of confusing.  `{"command":"set_white"}` should work as well.

sidoh, 2018-03-12:
This confusing behavior will be fixed in 1.7.0, but I'm taking a different approach.  Bulbs will just not respond to commands (or update state) while in night mode or in off.

closed 2018-05-05 by sidoh