sidoh/esp8266_milight_hub #172: color -> night -> color

servim, 2017-11-26:
1.6.2 RGBW
If you select a color after the night mode, then the lights first turn on the previous color, and then selected.

sidoh, 2017-11-26:
Unfortunately don't think there's a way around this.  Night mode is basically equivalent to "off."  The only commands the bulb responds to are on and off.  Sending other commands to the bulb works because they're paired with "on," and "on" is processed first.

servim, 2017-11-27:
just wanted to clarify, thanks!
And for the work done, thank you, good work!

sidoh, 2017-11-30:
Certainly possible to have a night mode button in the UI.  Obviously I'd welcome a PR! :)

Going to close this one since it seems like it's expected behavior, but feel free to reopen if I'm misunderstanding.

closed 2017-11-30 by sidoh