sidoh/esp8266_milight_hub #173: Add temperature and saturation controls and properties to RGBW bulbs

RvonMallesch, 2017-11-28:
Hi,

According to http://limitlessled.com/dev RGBW bulbs support adjusting saturation and kelvin. Would you be so kind as to add this control and add these properties plus color_temp properties? I'm particularly interested as I would like to implement functionality similar to f.lux.

Kind regards,

Richard

sidoh, 2017-11-29:
Not sure I'm following.  RGBW devices support a fixed temperature white LED and RGB.  You can't adjust the color temperature on these devices.

There are other devices (usually called "RGB+CCT" in limitlessled/milight marketing) that support adjustable color temperature, but those are already well supported by espMH.

RvonMallesch, 2017-11-29:
Maybe I'm missing some thing but if you look at the UDP commands V6 RGBW/WW/CW Commands

31 00 00 08 02 SS 00 00 00 = Saturation (SS hex values 0x00 to 0x64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)
31 00 00 08 03 BN 00 00 00 = BrightNess (BN hex values 0x00 to 0x64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)
31 00 00 08 05 KV 00 00 00 = Kelvin (KV hex values 0x00 to 0x64 : examples: 00 = 2700K (Warm White), 19 = 3650K, 32 = 4600K, 4B, = 5550K, 64 = 6500K (Cool White))

Those properties are listed. When I look at the box it talks about Kelvin range.

Anyway, as far as I'm concerned, I've found they make rgb_cct (or rgbww as they have been sometimes been labeled) bulbs in the few fitting types and sizes that didn't have that as option when I bought them originally. So I'm going to by them, so my whole house will have rgb_cct bulbs, BTW, all my rgb_cct bulbs responds to the fut089 commands. I've also found a way to control over 8 groups in fut089. If the group is over 8 add 12 to it. The only downside is that to turn them off you have to choose the all off button. This suits me fine as I use the device id for the room, and I assign a different zone for each bulb in the room (allowing for producing some cool effects). My room with the most lights has 9 so that was my reasoning behind my exploration. I'll do a push request when I'm finished.

Regards,

Richard

bombcheck, 2017-11-29:
The new RGB+CCT-bulbs support saturation control. It is very ugly because power output from the white leds is a lot higher than from the RGB ones. So as soon as saturation is below 100%, all gets messed up...
But the "old" RGBW bulbs did not have this feature. You could buy RGB+WW or RGB+CW bulbs. But there where bulbs on the market who did NOT have RGB but just white. These bulbs had adjustable color temperature.

sidoh, 2017-11-29:
Right, those UDP commands are for different devices.  espMH already supports them under the "RGB+CCT" (FUT092) or "FUT089" remote types.  RGBW devices do not support saturation control unless there's something hidden in the RF protocol. 

Ah, interesting.  I noticed a similar effect when I was first experimenting with FUT092 -- it can control 8 groups with the same tradeoffs.  Think it might be confusing to advertise that 12 groups are supported when it's 8 full-featured and 4 with unclear side effects.

sidoh, 2017-12-04:
@RvonMallesch, I'm going to close because it seems like what you're asking for is already supported, but please feel free to re-open if I'm missing something.

closed 2017-12-04 by sidoh