sidoh/esp8266_milight_hub #174: send and receive errors

balert, 2017-11-28:
Please mention on your wiki page or where else you consider appropriate that using a 10uuF capacitor between Vcc and Gnd of the NRF24 module can save you hours of debugging, as in my case light switching didn't work without it, but works perfectly like a charm with the capacitor in place. I originally read about this at http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo so it's actually not my own idea but I thought mentioning it could help others.

sidoh, 2017-11-28:
Interesting.  I've not needed this.  Perhaps it depends on the dev board and the power supply you're using?  I've also dismissed a bunch of NRF modules as junk, and maybe this is all they needed.  Will give it a try.

bombcheck, 2017-11-29:
Have those capacitor on my modules, too. Does not work to well without...
Here is some input: https://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo

sidoh, 2018-03-12:
Updated the troubleshooting guide:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting

thanks for pointing this out!

closed 2018-03-12 by sidoh