sidoh/esp8266_milight_hub #175: MQTT connection problem

nightcat91, 2017-12-01:
Hi,

I am using the newest version (1.6.2) uploaded on a NodeMCU V2 using PlatformIO.
Was able to contoll a RGBW bulb using the Web Interface and using an app on my phone, that was able to see a V5 bridge (it found it by itsself after I added a UDP bridge V5 in the web Interface).

Just the MQTT part gives me problems....
Using HomeAssistant (Hass.io) with the build in MQTT Broker (Mosquitto).
In the log from the MQTT broker I can see this message pop up about every 2 seconds:

1512120663: New connection from 192.168.10.170 on port 1883.
1512120663: Socket error on client <unknown>, disconnecting.

The IP adresses, ports and logins (username and passwort) are correct.

Any help would be very nice,
Thanks

sidoh, 2017-12-01:
Would be helpful to get serial logs from the ESP.  Ideally with the MQTT_DEBUG flag enabled.

HASS' MQTT broker doesn't behave according to MQTT specifications.  It's caused other issues (see #157), but I think that particular problem should be fixed.  Would recommend using Mosquitto if at all possible.

sidoh, 2017-12-04:
@nightcat91, are you able to provide Serial logs?

Another thought - do you have TLS enabled on port 1883?  espMH does not currently support MQTT with TLS.

nightcat91, 2017-12-06:
@sidoh, i will try to provide logs in a few days. 
Is that the correct place to enable the MQTT_DEBUG??
https://imgur.com/a/gQJRF

I am using the Mosquitto Broker that is installed as an addon to Hass.io
Dont have TLS activated, but I am using username and password to login. I checked to make sure that the login is correct.
What I noticed: I had the same error message pop up in my MQTT broker log when I had another ESP8266 Device trying to connect to the Broker but with the wrong login credentials.

sidoh, 2017-12-06:
Great. :)

Yes, that's the right place to enable the debug flag.  Just comment that line out.  May need to move it onto the same line, not sure how platformio's parsing works.

Gotcha.  I use username/password on my mosquitto broker as well, so that should work.

nightcat91, 2018-01-05:
Sorry about the long wait....
I wasnt able to do any debugging, I got caught up with other things sadly.

But I have good news.
I can confirm, that the problem is gone!
And since I didnt change anything software side about the milight hub (it is still the same ESP that I installed in on about 2 month ago), it must be something HomeAssistant changed with their MQTT broker.
I am now runnning HomeAssistant version 0.60 
The other software version along side it are the following: 
Host OS:
Type | resinos
OS | GNU/Linux
Host Control version | 1.1
Supervised Version: 0.78

Build in MQTT broker version: 1.0 on port 1883 with username and password protection

This setup right now is working very well and I was able to use the wiki and other good sources to set up listening and integrate my MiLights into HA using the "mqtt_json Light" Component.

Thanks again for your great software!

closed 2018-01-13 by nightcat91