sidoh/esp8266_milight_hub #176: Wake up alarm

starkillerOG, 2017-12-05:
I would love to use my Milight bulb on my bedroom ceiling as a wake-up light.
I already wrote some simple python script that can do this remotely if I keep it running on a raspberry pi (still in progress but sort of working).
But it would be much nicer if somthing like this could be implemented in the ESP_Milight_Hub itself.

Only not sure if it is visable to have the ESP checking the time against the set alarm(s) every 10 seconds or so. I don't know much about ESP's and am still learning, but from my understanding it does not support multitreading, so I think that means you can not just lanch a seperate while loop for checking the time besides all the other stuff the ESP Milight Hub does. Is that correct?

The python script I have now:

```

import requests
import json
import time

#=====================================================================================================================================
# Settings
#=====================================================================================================================================

# Settings
ESP_Milight_Hub_Ip = "192.168.1.???"
ESP_Milight_Hub_Username = '??????'
ESP_Milight_Hub_Pasword = '??????'
Bulb_ID = "0x????"
Bulb_type = "rgbw"
Bulb_group = "4"
debugging = 1
# Alarm Settings:
alarm_time = ["di, 10:23","11:40"]          # time the alarm goes off syntax: ["di, 10:34"] or ["10:34"] or a mix for multiple alarms such as ["di, 10:23","12:23"] 
rgb = [255,130,0]                           # sunset color

#=====================================================================================================================================
# Define the sunrise effect as wake-up alarm
#=====================================================================================================================================

def sunrise_effect():
    # command arguments
    url = """http://"""+ ESP_Milight_Hub_Ip + """/gateways/""" + Bulb_ID + "/" + Bulb_type + "/" + Bulb_group
    headers = {"content-type":["application/json"]}

    # sent color command
    payload = {"level":1, "status":"on", "color":{"r":rgb[0],"g":rgb[1],"b":rgb[2]}}
    r = requests.put(url, headers=headers, data=json.dumps(payload), auth=(ESP_Milight_Hub_Username, ESP_Milight_Hub_Pasword))
    if debugging == 1:
        print("brightness: 1, color: " + str(rgb))

    # brighten loop
    for step in range(2, 101):
        time.sleep(1)
        level = int(round(step))
        if debugging == 1:
            print("brightness: " + str(level))
        payload = {"level":level}
        r = requests.put(url, headers=headers, data=json.dumps(payload), auth=(ESP_Milight_Hub_Username, ESP_Milight_Hub_Pasword))

#=====================================================================================================================================
# Check the time in a loop if an alarm schould go off
#=====================================================================================================================================

count = 0
while (count<10):
    count = count + 1
    if debugging == 1:
        print(time.strftime("%a, %H:%M"))
    if (time.strftime("%a, %H:%M") in alarm_time or time.strftime("%H:%M") in alarm_time):
        if debugging == 1:
            print("alarm")
        sunrise_effect()
        break
    time.sleep(10)

#=====================================================================================================================================
# Diagnostics
#=====================================================================================================================================

if debugging == 1:
    print("done")
```

I am planning on implementing some MQTT in this python script to be able to easily set alarms etc.

But it would definetly be nicer to have this build into ESP milight Hub, is someone willing to help me implement this in the ESP milight Hub?

sidoh, 2017-12-05:
Nice script!

While I'm happy to entertain anything folks would find useful, I'm hesitant to add home automation functionalities into espMH.  I think there's tons of great software out there that really excels at this.  I use HomeAssistant for this kind of thing and love it.  HASS has an addon called AppDaemon that allows you to write python scripts.

starkillerOG, 2017-12-06:
I agree with you, I have been wanting a second Raspberry Pi for a while to install HASS (my first Pi runs Openelec which is not compatible with HASS).

closed 2017-12-06 by starkillerOG