sidoh/esp8266_milight_hub #177: Ambilight using Hyperion connection

starkillerOG, 2017-12-06:
Hyperion captures the images of your TV and transforms them to color commands for RGB Led strips behind your TV to create an ambilight behind your TV.
It also works together with various lamps like AtmoOrbs or Philips Hue lights to make them sync with the TV image to create extra ambilight effects.

I am hoping to connect Hyperion to the ESP milight hub so hyperion can sent the RGB/HSL values to my Milight bulbs.

My question is, which protocal is the fastest to sent RGB values to the ESPMH?
REST endpoint, MQTT or UDP?

What kind of response time can I expect for the fastest protocol and what kind of refresh rate?

I already have a RGB led stip that is directly connected to an ESP8266 (using a H801 controller which is just an ESP8266 with mosfets etc and a nice housing and connection ports for the RGB strip). This device (AtmoOrb) receives the RGB values over wifi using the UDP protocal. It has a response time in between 1 to 2 ms and can achieve about 60Hz refresh rate. It looks awsom as an extra ambilight besides the main ambilight behind my TV. However that ESP8266 is purely ment and optimized to receive UPD commands and can then directly control the RGB leds.

Has anyone tryed if the ESPMH can be fast enough to use as ambilight?

bombcheck, 2017-12-06:
I experimented a bit and tried to make an ambilight with my milight bulbs. It lags behind the picture and that is very anoying. Not to blame this great project for it, but the whole over-the-air-protocol is too laggy for that I think.

sidoh, 2017-12-06:
I've had most luck with MQTT.  You can try running through this:

https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning

But I would not expect anything even close to 60 Hz.  Maybe closer to 5-10 Hz is attainable.

There are some things that can be done to improve performance, but I've definitely not optimized for that.

Really fast throughput is theoretically possible (see @khmann's [project](https://github.com/khmann/milightx)), but there's a lot of work to get espMH even close to that level of performance.

starkillerOG, 2017-12-08:
Thank you for your response.
Lookes like at the moment it is not worth trying to connect Hyperion then.
Thanks for these tips, saves me a lot of time.
I will keep an eye on the MilightX project, if that works and gets of the ground maybe I will install that on a second controller to use for high throughput and keep ESPMH for nice reliable control and home automation.

closed 2020-03-28 by starkillerOG