sidoh/esp8266_milight_hub #178: NodeMCUv2 stops working after adding a device to a gateway server

mderoode, 2017-12-06:
First of all, I would like to mention that I really appreciatie the amount of good work which has been put into this project! This project was the inspiration for me to start experimenting with the Milights. 

I have a Milight RGBW which has never been used and a NodeMCUv2 hooked up to a NRF24L01+. I can control the light by using Device ID _0000_, I can change the color, brightness and set a disco mode 😄. This all seems fine, however, whenever I try to add this device to a gateway the NodeMCUv2 simply freezes and the webpage is not responding. Even if I reboot the device, the webpage won't work anymore (and according to my router the device is not connected to WiFi, moreover, the device does not broadcast it's default WiFi). The only 'fix' which seems to work for me is to erase the flash and reflash the firmware. I tried another power supply, but this also did not make any difference. Finally, I tried to switch the NRF24L01+  modules, but it unfortunately also did not change anything. 

Additionally, I do not seem to be able to pair the light. I tried setting several IDs (1,2,5, 0x0CD86) and hitting the pair button just after I turned on the light (within two seconds) and after several seconds/minutes. This has no effect whatsoever, although the sniffer does receive the send packages. I also tried to place the device very close to the light, but this has no effect (note that the control commands do work). 

Do you have any idea if I am doing something wrong? 

Thanks, 

Mike

sidoh, 2017-12-06:
Glad it's been useful.

Are you able to get serial logs?  Sounds like it's crashing.  Which version are you using?

Some bulbs are only able to be paired with one device ID.  If you haven't already, try unpairing it with the device it's paired with.

mderoode, 2017-12-06:
I used the latest 1.6.2 firmware. How do you enable serial logging? I don't get any log messages via serial input. 

I also tried the unpair button, but this has also no effect.

sidoh, 2017-12-06:
You shouldn't need to enable serial.  Baudrate is 9600.  Do you not see anything?  You should at least see some stuff during initialization.

> Edit: unpairing and then pairing did work so one of the two problems is already solved!

Sweet 👍

mderoode, 2017-12-07:
Ah sorry I used a high boudrate. When reproducing the problem the log shows nothing special: 

_*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.2.231
Setup complete_

When rebooting the device after it crashes, I get exactly the same log (see below). However, I now cannot use the web interface. 

_0It,ih⸮lE⸮lb8I⸮⸮*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.2.231
Setup complete_

My apologies for not giving too much helpful details.

sidoh, 2017-12-07:
If the log isn't showing anything, my guess is that it's spinning, not crashing.  When ESPs crash they spit out a stacktrace and an error code.

Could you clarify the exact steps you're following that cause this behavior? I can add a UDP gateway with the device ID 0000 just fine:

[image]

What are you setting the port to?

mderoode, 2017-12-07:
First, I set the Device ID to 1 and use some controls. Then I enter in the Gateway section the Device ID and set a UDP Port. I tried ports ranging from 10 to 9001 to 5655. Finally, I simply hit the save button and then the device doesn't respond anymore (the page doesn't give an error, but no controls work and reloading the page results in a 'Can't reach this page'-error).

sidoh, 2017-12-07:
Could you paste the requests (ideally as curl commands) that your browser is sending when you do this?

example, Chrome on OS X:

[image]

```
curl 'http://10.133.8.158/settings' -X PUT -H 'Pragma: no-cache' -H 'Origin: http://10.133.8.158' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: en-US,en;q=0.9' -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/62.0.3202.94 Safari/537.36' -H 'Content-Type: application/json' -H 'Accept: */*' -H 'Cache-Control: no-cache' -H 'X-Requested-With: XMLHttpRequest' -H 'Connection: keep-alive' -H 'Referer: http://10.133.8.158/' --data-binary '{"gateway_configs":[]}' --compressed
```

mderoode, 2017-12-08:
The cURL command is (Chrome on Windows):

> curl 'http://192.168.2.231/settings' -X PUT -H 'Origin: http://192.168.2.231' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: en-US,en;q=0.9' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/62.0.3202.94 Safari/537.36' -H 'Content-Type: application/json' -H 'Accept: */*' -H 'Referer: http://192.168.2.231/' -H 'X-Requested-With: XMLHttpRequest' -H 'Connection: keep-alive' --data-binary '{"gateway_configs":[["1","9001",5]]}' --compressed

sidoh, 2017-12-08:
I'm still not able to reproduce it.  Could you paste the output of:

* http://192.168.2.231/settings (make sure to remove any passwords)
* http://192.168.2.231/about

mderoode, 2017-12-09:
I reflashed the same firmware (esp8266_milight_hub_nodemcuv2-1.6.2.bin) again and this is the content of settings:

> {"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"group_state_fields":["state","brightness","computed_color","mode","color_temp","bulb_mode"]}

About:

> {"version":"1.6.2","variant":"nodemcuv2","free_heap":23800,"arduino_version":"00000000","reset_reason":"External System"}

After performing the save operation, I cannot access the device anymore (and hence not read the /settings and /about page).

sidoh, 2017-12-09:
Are you able to compile with the `UDP_DEBUG` and see if there's anything interesting in serial logs?  This is still not happening on my espMH install :(

mderoode, 2017-12-13:
I compiled from source code and uploaded via the web form and suddenly the device seems to work. I am now able to add gateway severs! I attached the firmware because it might be useful for you (perhaps something went wrong during compilation of the latest release?) .Please note that this is the nodemcuv2 firmware. 

Thanks!

[firmware.zip](https://github.com/sidoh/esp8266_milight_hub/files/1554794/firmware.zip)

sidoh, 2017-12-13:
Hmm... I'm using the precompiled releases as well.  It is really strange that your binary is ~20 KB larger.

How were you flashing the firmware?

mderoode, 2017-12-13:
That's weird, I downloaded Atom and installed PlatformIO. Next, I just set the UDP_DEBUG flag and compiled. For some reason, I was not able to flash the NodeMCUv2 via serial (I could read the serial output, but esptool couldn't find it), therefore I tried uploading the firmware via the web interface of device (as it was currently running an other version of your software).

sidoh, 2018-03-12:
It sounds like you got this working, so I'm going to close the issue.  But please feel free to open another if you're still having problems.

closed 2018-03-12 by sidoh