sidoh/esp8266_milight_hub #179: name of the ESP module

servim, 2017-12-07:
Hi! How to change the name of the ESP module?
It is not informative on the network to see the name ESP_3C6896.
Perhaps this could be done at the time of the initial setup, but I missed it. If so, how do you get back to the initial setup? Thank you!

mderoode, 2017-12-07:
I think the name is hardcoded in the source code and corresponds to `ESP.getChipId())`. Look at https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp at line 260. You should be able to change it manually and recompile from source.

sidoh, 2017-12-07:
@mderoode is essentially right, the hostname isn't currently configurable.  [main.cpp#260](https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp#L260) is actually setting the name for SSDP.  I could be wrong, I'm guessing that @servim is talking about the hostname.

Would just need to call `ESP.hostname(String)` to set the hostname set along with a DCHP request.  Should be pretty straightforward to expose this as a setting.

servim, 2017-12-07:
Unfortunately I did not understand everything that you are saying, but the essence is clear to me! you must specify the name yourself and recompile. True, I have never seen platformio, but in Arduino it probably does not compile. Put in the plan, get acquainted with the platformio. Thank you!

sidoh, 2017-12-07:
Yes, just call `WiFi.hostname("your-host-here");` _before_ `wifiManager.begin()`.

servim, 2017-12-07:
sorry, I do not understand how and where it should be written :)

sidoh, 2017-12-07:
here should work:

https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp#L243

servim, 2017-12-10:
Yes, I understood this :)
Only I can not compile. I do not know platformio. I will sort this out...

Skinah, 2018-02-03:
I have an Asus router with the merlin firmware loaded and you can set hostnames and lock them to the mac address of devices on the wifi. No need to compile this if you have the ability in your router.

rjsachse, 2018-03-17:
Yes it would be great to have a setting to change the name. At least change ssdp from esp-chip id to MilightHub-chip id

closed 2019-05-28 by sidoh