sidoh/esp8266_milight_hub #180: ESP-201 saving settings not working

VincentGijsen, 2017-12-09:
I have flashed my esp-201 board with a released fw, i think currently ive insalled the ESP12 variant.

COnfiguration of the wifi works well, but I cannot save any of the changes on the milight dashboard, create a server or anything, also I cannot make the change to LT900 perisitent either.

Do i need to roll my own fw? i know there is the 512kb flash chip, I already changed DIO/QIO changes, but seems to not change anything.

Also downloading of the (empty) config yields network exception on the download. I use Chrome btw. but as it seems bootstrap, my guess is it shouldnt matter).

any thoughts, could this be related to the board-specific fw's, and how do they differ, i would guess only by flash size and perhaps a crystal?

sidoh, 2017-12-09:
As you were hinting at, the only substantial differences in the current pre-built images are the flash boundaries that separate the code region from the FS.  Currently they all assume at least 1 MB of flash.  

All of the symptoms you're listing are, I think, consistent with SPIFFS not being able to write anything.

Judging from an article linked in #145, the firmware image is too large to fit on a 512 KB chip.

VincentGijsen, 2017-12-09:
Hi Sido,

thanks for your comment, in it's light, added the following lines to the platform.ini

```

[env:esp201]
platform = ${common.platform}
framework = ${common.framework}
board = esp07
build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.512k64.ld -D FIRMWARE_VARIANT=esp07
extra_scripts = ${common.extra_scripts}
lib_deps =
  ${common.lib_deps_builtin}
  ${common.lib_deps_external}
```
and run it with the -e esp201

it works, not sure if all is well, but it does save data now ;)

sidoh, 2017-12-09:
Oh, nice!  Glad it's working.  Would be happy to add esp201 to the list of pre-built images.

I'm guessing that OTA updates won't work with 512 KB flash, but as long as SPIFFS is working there's probably no other disadvantage :)

VincentGijsen, 2017-12-10:
well i guess i celebrated to soon.... it works a single-run, that is, after wifi-setup, the device registers on my own ssid. I can send commands to the lights, but a power-cycle later.. it doesn't start anymore, and my router shows me that the device doesn't connect anymore to the wifi (so associate, not dhcp req). 

I guess only thing to do, is either 

* buy nodemcu
* replace spi-flash

too bad must throw-away my soldered proto-board ;(

or do you have any idea's how to limit the binary to 423 ;P

sidoh, 2017-12-10:
Ah, bummer.  

You could fork and trim stuff that you don't need.  Trimming UDP stuff would probably save a fair amount, but not sure how far it'll get you.

sidoh, 2017-12-10:
I'm going to close this since it sounds like the mystery is resolved, but feel free to re-open if I'm missing something.

closed 2017-12-10 by sidoh