sidoh/esp8266_milight_hub #181: How to control 4+ groups on the webadmin and API?

markg85, 2017-12-10:
Hi,

I'm sure this is a silly question as the project states quite clearly that it supports well above 4 groups.
Yet in the screenshots of the webadmin I see a 4 group example.
In mqtt code I also see 4 groups popping back up.

That kinda makes me wonder, how can I control or even assign group - let's say - 10?
Is that just visible on the admin page if there is a group 10?
Can I just do a mqtt call to a group 10 as I can also do with the first 4 groups?

This probably would be clearly visible to me if I had this running, but I haven't yet. The components for it are ordered though :)

Note: I'm sure the advice is possible, but could you please update the screenshots to show this awesome 4+ groups feature?

Cheers,
Mark

sidoh, 2017-12-10:
You can emulate as many device IDs as you want from a single install of espMH.  The limit of 4-8 groups per device ID is imposed by the RF protocol, but there's not a ton of advantage to pairing many devices to the same device ID.

Most remote types support four groups.  FUT089 supports 8 groups.

markg85, 2017-12-10:
Hi,

I'm not quite sure if i understand.
I personally have a use for about 16 groups:
- Living room
- Kitchen isle
- Kitchen backlight
- Ceiling corner light (living room)
- Entrance
- Bathroom
- Shower
- Laundry room
- Back garden lights
- Front garden lights
- etc...

I left a couple out, but i can think of at least 4 more "groups". Each group could have 1 or more light. 

How can i do that with your firmware?
I just don't see it... What do you mean by pairing devices to a device id? You can pair them to something other then a group? I don't see that in the GUI screenshots..

Personally, i don't mind at all if that would not be compatible with the milight app as i've been using that for about a year now with great frustration. It is very crash happy... As long as it can be made to work with Smartthings, which i think it can.

sidoh, 2017-12-10:
Device ID is a 16 bit identifier usually bound to a physical milight device (remote, wifi gateway, etc.).  With espMH, you can set this ID dynamically in a number of ways.  In the UI, it's here:

[image]

In the [REST API](https://github.com/sidoh/esp8266_milight_hub#rest-endpoints), it's part of the URL, and so on.

@fireboy1919 wrote up some great instructions on integrating with SmartThings:

https://community.smartthings.com/t/release-new-milight-easybulb-limitlessled-integration/84925

markg85, 2017-12-10:
Am i correct in assuming that in your terminology the "Device ID" is your alternative for groups? And groups is merely there to be compatible with MiLight, but can be ignored? In other terms, A device id can be linked to any number of devices and there can be a unlimited (roughly..) number of device id's?

If that's how i need to interpret it then it's all clear to me :)

sidoh, 2017-12-10:
That's mostly right.  Only thing I'd add is that Device ID is something that exists with official Milight devices as well.  With the official devices, though, they're hardcoded. 

The way I think about it is that bulbs are linked to a _(device\_id, group\_id)_ pair.  With the official devices you can only control _group\_id_, but with espMH you can set _device\_id_ to whatever you want as well.

Device IDs are 16 bits, so you're limited to 65536 values, but if you have that many groups to control you'll probably run into other problems =)

markg85, 2017-12-10:
hahaha, remembering that many groups is going to be a challenge ;)
Anyhow, thank you for the clarification!

I was having a hard time finding that.. You might want to update the README.md to reflect this, as it's quite a big advantage over the stock MiLight capabilities.

sidoh, 2017-12-10:
I'd meant for it to be addressed in (1)  [here](https://github.com/sidoh/esp8266_milight_hub#why-this-is-useful).  Doesn't go into a ton of detail, though.

closed 2017-12-10 by markg85

markg85, 2019-12-29:
Hi,

Sorry for necrobumping this topic, but as i made it in the first place (and closed it) and now want to thank you, this might as well be the best place.

Thank you so much for this awesome project!

Back when i wrote the above i was playing with milight and integrating it in my home. I eventually did not use this project.

Now, a few years later, i still have the very same milight and am increasingly plagues by lights that will only turn off after 10x on/off toggle. The gataway seems to be dying on me i suppose. So now i did finally solder an ESP8266 (12-E) and an nrf24 together. The only issue i had was GPIO15 which i had tied to ground before adding the nrf chip. Now it's pulled down to make it all still work. Some firmware flashing and configuring followed...

Fast forward a few hours...

And i'm totally amazed by the rapid responsiveness of your project and my lights! No more seconds waiting till the lights go on/off. It happens as soon as i press the magic button in the web ui! Finally more then 4 groups! Finally a way better integration with google assistant (integrating it through your mqtt protocol and some smartthings frustrations). 

Besides smartthings, i'm totally happy about this :)
A very nice addition that could make it even more great is if a local google assistant gateway were to exist for it. That's possible since the last Google I/O and keeps the snappy feeling as turning a device on/off is then done fully locally. Still, the command you tell to the assistant is being send to the internet so a bit of latency does remain.