sidoh/esp8266_milight_hub #184: New feature: Repeater-Mode

bombcheck, 2017-12-14:
Hi,

it would be great to have a "repeater mode", where the system sniffs for packets from another hub and re-sends them. Could be useful if you have a wide area to cover. :)

janLo, 2017-12-14:
I'm not sure if this is a good idea with only one radio. It adds latency and because of the frequency hopping you're at risk to miss packets. Maybe just group you lamps to different hubs and give them disjoint mqtt topics. So you only need to make sure, that the wifi coverage reaches all hubs and you're fine.

sidoh, 2017-12-14:
I've been thinking about something like this as well.  I don't know enough about the nRF to say for sure, but my feeling is that janLo is right, a straight RF repeater would likely be unreliable.  But the feature would be really useful, I think.

There are some other approaches I've been thinking about:

1. Several espMHs listening on the same MQTT topic.  Could disable state updates from all but one of them.  Repeat packets shouldn't be a problem except in some edge cases where packets don't define a state (like "set brightness to 50%"), but are instead deltas ("increase mode speed by 1").

2. Same as (1), but include a packet ID field in the command packet.  This could be pushed into the RF packets, and would solve the duplicate problem.  Adds obnoxious moving pieces.

3. Similar to the original idea, have master/slave mode.  Would be similar to (2), except the ID field would be managed by the master.

janLo's suggestion is good as well, and has the benefit of not requiring any changes.  :)

Tertiush, 2017-12-14:
Just a though; what about a "master" hub broadcasting the command on your LAN (on its broadcast address) on a specific port using UDP. All ESPs listen on this port and simply repeats, perhaps with some selectable delay.....

khmann, 2017-12-18:
Make it listen to multicast address!  Then it goes everywhere automatically...  Disable frequency hopping so there are no collisions.  For me, "repeater" is the same as "macro controller" so I'm researching this idea

sidoh, 2017-12-18:
I guess MQTT feels better to me than a custom UDP multicast protocol because:

1. Its structure already lends itself pretty nicely to the requirements.
2. The protocol includes security/authentication mechanisms.
3. It's a lot less work that cooking up a custom protocol :)

Only real downside is that it introduces a dependency on having an MQTT broker set up.

sidoh, 2017-12-18:
@khmann you've certainly convinced me that it's silly to have the controller frequency hop.  I can experiment with disabling this after the new year.  Would you expect it to be useful to expose a way to select which channel is getting used?  Or would it be mostly safe to just select one arbitrarily?

khmann, 2017-12-19:
ha.  It's easy to criticize, especially when you don't stick around to help  ;)

One of the things I really like about your project is that it's a functional iBox "Reference Implementation".  So perhaps fixed-frequency should be optional, as part of a 4-way dropdown next to retries: Hop/Bottom/Middle/Top, Auto/A/B/C, Diversity/L/M/H...?  In the FCC, the high frequencies are much farther from 802.11 networks (10-11 MHz instead of 1-2 MHz (RGB[W,CCT])and so are the safe default, followed by the low ones because the "middle of the band" is always a gamble.  If you were super-crafty, you could auto-select Lo/Hi whichever was farther from the WiFi channel you were using.

re: MQTT vs UDP sorry, I guess I had only ever considered the native milight UDP protocol - that one works out of the box if you just advertise with the broadcast MAC address.  How you'd do it effectively with TCP MQTT stuff, I'm not even qualified to speculate

RvonMallesch, 2018-04-26:
I can't up vote this enough. Do you think it is possible to make it into 1.7.0?

sidoh, 2018-04-27:
I think all of the major features that will be in 1.7 have already been merged.  Just working on getting it stable.

Open to including this in the next major release. 

There are a couple of ideas being discussed in this ticket.  Which are you referring to, @RvonMallesch?  Repeater mode?

Is there a reason you couldn't just set up multiple ESPs listening on the same MQTT command topic?  I thought about this earlier, and feel like there was some problem, but I'm not able to recall what it was.

sidoh, 2018-04-28:
Definitely agree the channel settings is a nice feature.  I'll prioritize it for the next major release.

RvonMallesch, 2018-04-29:
Sorry for the late reply, it as the channel settings, I was referring my support for. I'm glad you agree! :-)

janLo, 2018-04-29:
In the end you've 2^16 possible groups. So I think, with a bit of planning it shouldn't be necessary to have more than one controller listening to the same topic.