sidoh/esp8266_milight_hub #185: Domoticz

joshdinsdale, 2017-12-15:
Hi All

Im back following this project again after some absence due to family and work stuff, the usual! Anyway, all of my previous issues now seem to have been addresses in the 1.6 branch which is great!

I'd like to know if anyone has had any luck getting this working in Domoticz by emulating a controller? I've tried adding as each remote type as both version 5 and 6 and i get nothing.

Is it possible to get logging in regards to devices accessing the emulated controller(S)?

joshdinsdale, 2017-12-15:
Ignore me, I have it working!

closed 2017-12-15 by joshdinsdale

alexsahka, 2019-03-06:
Could you help me? I can't figure out how is this HUB emulating a controller in the Domoticz.
If you can take any screenshots for me at least.