sidoh/esp8266_milight_hub #186: FUT07 Remote + MQTT

joshdinsdale, 2017-12-15:
Hi All, this isnt really an issue more of a request for advice...

I want to use a FUT07 remote via the ESP to send MQTT messages to my domoticz controller. This will then send commands back to the ESP to control my milight LED strips.  As I understand it, this will allow me to keep state between the remote and domoticz.

This is my first time in even looking at MQTT so I'm not exactly sure what I'm doing! I've done mosquitto_pub tests against my broker and can see the results when i do mosquitto_sub. However I have set the ESP to send to the same topic, but whatever I do i cant seem to see anything when debugging with mosquitto_sub.

Does anyone have any pointers? Is what I'm trying to achieve even feasible???

Thanks.

Josh

sidoh, 2017-12-16:
Should be possible.  Here's a howto on doing the same thing with HomeAssistant:

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

joshdinsdale, 2017-12-16:
I've now managed to tune into the messages from the milight controller. The issue I am now facing is parsing the data into a form that domoticz can understand. I have node red installed and I assume this would be the best way to parse the data?
Anyway I guess this isnt your issue as the controller works 😁

sidoh, 2018-01-04:
Hey @joshdinsdale, sorry for the slow response.  I'm not familiar with NodeRed or Domoticz, but from what I do know I'd imagine it'd be possible.  

Anyway, going to close this since it sounds like your issue is fixed, but feel free to reopen.

closed 2018-01-04 by sidoh