sidoh/esp8266_milight_hub #187: Sniffing problem

CaCu15, 2017-12-18:
Greetings Chris,

I'm just trying your Milight HUB installed on the NodeMCUV2, but I'm facing a problem with the sniffing function.  I'm able to access to web UI of the milight hub, but if I start to sniff traffic using the button "Start Sniffing" and press some buttons on the milight remote I can see the LED on the NodeMCU flashing but
the UI doesn't show any captured messages. I checked my wiring against the description on http://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/, but everything seems OK.  Do you have any ideas how to solve this problem?

Thx in advance for your help,

Carsten

sidoh, 2017-12-18:
Hi Carsten,

Check out this troubleshooting guide:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting

It's possible it's bad hardware.  

If you're pretty sure the hardware's good, you could compile the firmware with the `DEBUG_PRINTF` flag and see if there's anything interesting in Serial logs.

CaCu15, 2017-12-19:
Hi Chris,

thx for your support. 

I tried 3 different NodeMCUV2s and 3 different NRF24L01+ modules, all are showing the same behaviour. 
Additionaly I compiled from source (latest version fom the repository, 1.6.3 (nodemcuv2) ) using the DEBUG_PRINTF flag. The serial console shows successfull reception and sending of packages. Using the output on the serial console I was able to find the device ID of my gateway. Using this gateway ID I was able to control my milight devices from the web UI, but sniffing is still not working - there's no output of packages in the web UI although they are shown in the serial console. 

But anyway: The main feature is working now. Sniffing will only be required in case I'm buying some new equipment. But I using the serial console to find the device ID of the gateway works as well.

Thx for your great work, I'm really looking forward to test this in combination with the diyHUE project ([https://github.com/mariusmotea/diyHue])

Carsten

sidoh, 2017-12-19:
Gotcha, thanks for trying different hardware.

If stuff is showing up in Serial, it should probably make it to the web UI.  Do you see anything interesting in your browser's dev console?  Could you paste some of the Serial logs?

CaCu15, 2017-12-19:
Hi Chris,

here is a small output from the serial console. It show usage of the buttons "All Groups On" and "All groups off" on a milight remote. 
[Serial_Log.txt](https://github.com/sidoh/esp8266_milight_hub/files/1570813/Serial_Log.txt)

Additionally I found the following error  in the FireFox dev console. First message is in German, it says "Firefox can not establish a connection to the server ws://192.168.4.95:81"

```
Firefox kann keine Verbindung zu dem Server unter ws://192.168.4.95:81/ aufbauen.
192.168.4.95:3:3211
Source-Map-Fehler: TypeError: NetworkError when attempting to fetch resource.
Ressourcen-Adresse: moz-extension://35342e24-6c40-45b8-95ab-c87a04cf0915/common/common.js
Source-Map-Adresse: common.js.map
```

sidoh, 2017-12-19:
Ah, that's definitely the problem.  Looks like it's failing to make a connection to the websocket.  Sniffed packets are pushed over the websocket connection.

Seems to be working on Firefox v57 for me, so all I can figure is a security/privacy setting.  Are you able to try a different browser just to confirm that's the issue?

CaCu15, 2017-12-19:
I found the problem: My browser is in a different subnet than the NodeMCU device and my firewall was blocking port 81 between these subnets.  Because I don't wnat to have all this cheap smart home devices in my main LAN I created a separate LAN for my smart home devices with very strict firewall rules between them - obviously a little bit to strict. After I opened that port on the firewall everything works fine now.

Sorry for bothering you, should have figured that out on my own. Thx again for your support!

Carsten

sidoh, 2017-12-19:
Glad you got it working!

UI could definitely produce some more informative errors when the websocket connection fails.

Mind if I hijack this ticket to log that improvement?

sidoh, 2017-12-19:
Actually probably clearer to create a new ticket that references this one.  Just did that.

closed 2017-12-19 by sidoh