sidoh/esp8266_milight_hub #188: MQTT group state fields

starkillerOG, 2017-12-18:
The Json blob used to contain the following keys with values:
- button_id
- device_id
- group_id
- command

I have upgraded to the latest version of the ESP milight hub but now the MQTT part has changed and those keys are not accesible anymore.
Those can also not be selected in the MQTT group state fields.
I used those keys for my custom Hyperion remote and that is now broken because i don't have those keys any more.

Could someone please please please add those keys back as an option for the "mqtt_state_topic_pattern"?
That would solve my proble.

sidoh, 2017-12-18:
The reason it was removed is because all of the same information is encoded in the topic messages are published to, so it felt redundant.  Not much harm in exposing it as an option, though.

Think it'd have to happen here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/BulbStateUpdater.cpp#L40-L55

I can take a look when I have some time, but might not be for a few weeks.  Happy to look at a PR if someone else wants to take a stab.

starkillerOG, 2017-12-19:
I looked and it seems to me the new keys need to be added at least:
[https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Types/GroupStateField.h](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Types/GroupStateField.h)
[https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLightState/GroupState.h](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLightState/GroupState.h)

But unfortunately I don't understand much of the code.

Would be greath if someone would have time to help me out!

sidoh, 2017-12-19:
You would need to add fields in the first link:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Types/GroupStateField.h

but not the second.  The second is for state associated with a BulbId, not the fields in BulbId themselves.  Should just be adding the fields and adding the handlers for those fields in BulbStateUpdater.

Think there's a copy of the field names in script.js as well.

I should be able to take a look after the holidays.  Won't be around much until then :)

starkillerOG, 2017-12-19:
I just found that v1.6.1 vs v1.6.0 is adding the computed_color to the group state field:
[https://github.com/sidoh/esp8266_milight_hub/compare/v1.6.0...1.6.1](https://github.com/sidoh/esp8266_milight_hub/compare/v1.6.0...1.6.1)
That schould help as an example how to add a group state field and wich lines should be altered.

Thanks so much sidoh, I really apriciate your help.
If I can assist in any way just ask!

sidoh, 2018-03-12:
This will be added in 1.7.

starkillerOG, 2018-03-12:
Thanks @sidoh

closed 2018-05-05 by sidoh