sidoh/esp8266_milight_hub #190: I do not see any traffic on my milight hub

rolfberkenbosch, 2017-12-19:
Hi All,

I have a problem and can't figured out what I'm doing wrong. I still not see any traffic when i press the start sniffing button. I have tried the following:
1. Recheck and recheck the cable wirring
2. Reimage the Wemos D1 mini v2
3. Reboot the wemos D1 many times.
4. Put the milight off and on many times.

I have added some photo's, maybe you guys can find my BIG mistake that cost me many hours.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sidoh, 2017-12-20:
If you haven't already, take a look at the [troubleshooting guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting).

Something that's not in the guide -- some folks (see #174) have mentioned that a 10µF capacitor across Vin and Gnd on the NRF24 can make a big difference.  I've not been able to verify this myself.

bombcheck, 2017-12-20:
As far as I can see, you are using a transceiver with integrated power amplifier. These models need an extra power source as the NodeMCU does not provide enough power for them.
https://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo

rolfberkenbosch, 2017-12-20:
I have tried an otter module without external antenne. Do i need a remote control to find the device id? I just put some  power on the bulb.

sidoh, 2017-12-20:
Ignoring some very weird corner cases (see #91), bulbs don't send traffic.

If you're just looking to get a new bulb working with espMH, pick whatever device ID you like (between 0x0000 and 0xFFFF) and pair the bulb with it.  (To do this, press the "pair" button in the UI within 5s of supplying power to the bulb).

rolfberkenbosch, 2017-12-21:
Thanks @sidoh that was the trick! The module works perfect!

closed 2017-12-21 by rolfberkenbosch

mames13, 2023-01-16:
Hi, I might be having this problem with a second Nodemcuv2 that I just built with the external antenna transceiver similar to what is pictured.  What voltage should the 10µF capacitor be? Thanks!