sidoh/esp8266_milight_hub #191: brightness sync problem in home assistant (mqtt)

badstraw360, 2017-12-23:
Im having a sync problem with home assistant and I'm not sure if it's a bug... 

I'm using a FUT012 bulb... So when I change the brightness to let's say 40% and then turn it off and turn it back on , the brightness of that bulb physically remains the same but the slider in home assistant will reset to 100%...
So the main issue is that I need to manually move the slider down and turn it up to reset the brightness... (im using mqtt)

Is there any possibility to sync the bulbs with homeassistant?

sidoh, 2017-12-24:
Could you include some MQTT logs?

HASS should stay in sync.  It does with my setup.

badstraw360, 2017-12-28:
It's working now, I don't know if it was like a glitch or something... I'll update the post if anything strange happens.

Thank you very much.

sidoh, 2018-01-03:
Thanks @badstraw360!  Feel free to re-open if it happens again.

closed 2018-01-03 by sidoh

badstraw360, 2018-02-26:
Ok I'm still having the same problem...

When I set the brightness to 40% and then close the browser, wait a few seconds and open it again, the brightness slider will reset to 100% but the bulb physically stays at 40%. So if I want to fix that I need to manually move the slider to update the bulbs again.

I'm using hassbian 0.64.0

I would appreciate your help. :)

```2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/2
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x5555/rgb_cct/2
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/4
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x5555/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x1111/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x0/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x2222/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x9999/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to /blind/bc/state
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x5555/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/4: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x5555/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x1111/rgb_cct/0: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x0/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x2222/rgb_cct/0: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x9999/rgb_cct/0: b'{"state":"OFF","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:43 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on stat/sonoff/POWER: b'OFF'
2018-02-26 09:55:28 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:55:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:03 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:06 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":48,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:08 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":25,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:19 ERROR (MainThread) [homeassistant.core] Timer got out of sync. Resetting
2018-02-26 09:58:24 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":80,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:25 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:26 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
```