sidoh/esp8266_milight_hub #192: radio IC has changed, RGB+CCT encryption revisited

khmann, 2017-12-26:
I didn't really know where to post this, because it's not an esp8266_milight_hub issue, but since Chris cracked the encryption in the first place... close examination of all the MiLight RGB+CCT enabled devices I have at my disposal indicates the PL1167 is no more.

I noticed this when I went to attach my new $16 logic analyzer to my WiFi iBox (I wanted to measure the "resend" and frequency hopping timing precisely)... the antenna was connected where I was expecting an SPI pin!  The visual queue is the new devices have the antenna connected to (in my case, unlabeled) IC on pin 10 versus pin 16 on older devices.

Perhaps if we can identify the chip we can find out more about the algorithm.  Can anyone read theirs?  Many of the bulbs (and the strip controllers) are super-easy to open non-destructively...

This could certainly explain how they are able to achieve flawless performance with a 16MHz MCU... "scrambling" done in the radio!

khmann, 2018-01-02:
Ok, it seems the LT8900 is just a flipped around PL1167, and thats what the IC in the 2016 iBox works like.  The SPI clock is less than 1 MHz so it's easy to sniff, I only needed to tap the top two pins 1 (MOSI) and 16 (CLK) and sigrok/PulseView did the rest.

Regrettably the communication in the RGBCCT SPI messages is still encrypted, as in the air.  I'm sure that's as-expected... I've since found some pictures online that show 1st generation FUT015s did in fact contain PL1189s, I guess the LT8900 redesign happened across the board at a later point... it's common across all my 2016+ stuff which is what started me on this "issue"

BUT I'm sure you've been waiting long enough to have the settings for the FUT020, FUT-021, FUT 022 vintage strip controllers:

syncword 0x50A0AA55 at 2478, 2408, 2443MHz... so there's that.  I'm going to hook a few more devices and try to present the information is a reasonable format.

sidoh, 2018-01-02:
Interesting.  This suggests they're doing the scrambling in software?

Very helpful to get the RF configs for other devices.  Thanks very much for digging into this :)

khmann, 2018-01-07:
Thanks for the encouragement ;) 

> Interesting. This suggests they're doing the scrambling in software?
Yeah, but you already knew that...  I was just hoping for something else.  I've now ordered some STM8 debug hardware to see if I can dump anything out via the SWIM port.  Your reverse-engineering is outstanding, but the implementation is too much for my controller... I need something simpler.

I think I just bought an FUT020 off amazon so I'll test that RF config and fork a protocol handler for the sake of completeness.  I also have a couple "fake" MiLight RGB controllers now (boy they _really_ suck), so maybe I'll do a handler for those also - to help those poor souls like myself who got scammed.

sidoh, 2019-05-20:
@khmann, cleaning out old tickets.  Gonna close this one, but let me know if I've missed something.

closed 2019-05-20 by sidoh