sidoh/esp8266_milight_hub #194: RGB+CCT White mode - color temp

starkillerOG, 2018-01-03:
I noticed that with my milight GU10 FUT103 the "white mode" does not behave properly with the RGB+CCT protecol.
When I have set my bulbs to warm white (color tempreture = 100) and I then place them in RGB mode and then press the "White mode" button in ESPMH, it set the bulbs to cold white (color tempreture = 0).
This is quite annoying.

However in the RGBW protecol this is not the case, there when I press the "White mode" button in ESPMH it turns back nicely to white mode with the previous color tempreture.

When I press white mode on my milight RGBW remote or B8 wall panel, both turn to the previous color tempreture as I would expect and is the case for the RGBW protecol in ESPMH.

Is it possible to fix this bug in the RGB+CCT protecol?

sidoh, 2018-01-03:
I agree that this is not ideal.

RGBW and the FUT089 (B8) protocol both have RF commands to the effect of "set bulb to white mode," which is why they behave like you would expect.

As far as I'm aware, FUT092 (RGB+CCT) does not.  If there is, I'd happily update the code to use it.  The only way to enter white mode is to set color temperature to some value.  I arbitrarily chose 0 -- could easily change the default to 100.  Most people probably would prefer that.

If nothing else, ideally it would set the white value to the last known state.  There are other places where this would be useful, so it'll probably happen eventually.

starkillerOG, 2018-01-04:
@sidoh Thanks for the explanation!
For me changing the default color temp to 100 would solve my problem.

But I agree that setting it to the last color temp would be the best solution.
With my ESPMH the color temp is remembered correctly as 100 but it switches to 0 when i press the "white mode" button. Therfore it seems that the color temperature is already correctly statefull.
The only thing that needs to happen then is use that value already stored in memory during the setting of the "white mode".

starkillerOG, 2018-01-04:
I think the only line that needs to be changed is:

[/lib/MiLight/RgbCctPacketFormatter.cpp line 58](https://github.com/sidoh/esp8266_milight_hub/blob/3b9c3cbd62a34f431df296d153ead654ae7405e6/lib/MiLight/RgbCctPacketFormatter.cpp#L58)

I just created a pull request for this change.
@sidoh could you check it and then accept the pull request?

starkillerOG, 2018-01-04:
#196

closed 2018-01-04 by starkillerOG