sidoh/esp8266_milight_hub #195: get device id

druckgott, 2018-01-03:
Hallo,

is it posible to implement to get the device id without an real conroller. So maybe a scanner?

sidoh, 2018-01-03:
Controllers are the things that have device IDs.  If you have bulbs or other lighting devices, you can choose a device ID arbitrarily and pair it with espMH.  Instructions here:

https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs

druckgott, 2018-01-04:
Ah OK on the mobile side I did not see this wiki, maybe you can link this on  the Readme page.

Maybe it will also be cool to have a generate button for this ID then it automatically explains that its defined what you want

sidoh, 2018-01-05:
Haha, it wouldn't have been there!  I wrote that article after you asked the question.  A couple others have asked it as well, so figured it made sense to write it up. :)

Hints in the UI sound like a good idea.  Honestly the UI pretty desperately needs an overhaul anyway.  Feel free to open another ticket for that.

closed 2018-01-05 by sidoh

druckgott, 2018-01-05:
Perfekt I se also this topics in the issue after I did post it 🐵