sidoh/esp8266_milight_hub #202: Crazyradio PA: nRF24LU1+ based USB dongle transciever with amplifier

SmilyOrg, 2018-01-09:
Just found this [Crazyradio PA dongle](https://www.bitcraze.io/crazyradio-pa/) and thought it might be a neat self-contained way to control lights for an entire neighborhood! 😄 

It does feature open firmware that you can update... so I bet with enough work you could plug it in a RPi and have it work with no hardware hacking. 😃

sidoh, 2018-01-10:
This looks cool, but is probably out of scope for this project.

It probably wouldn't be a crazy amount of work to port the packet handling stuff to another project, but I'm not sure it makes sense to support this device within the same project.

SmilyOrg, 2018-01-10:
Yeah, I agree. Linked it for relevance :)

closed 2018-01-11 by sidoh